mbeddr/mbeddr.core #1593: Message "variable may not be initialized!" has error severity bothering development with mbeddr

huibo-shi, 2016-12-05:
Hi,

Please have a look at the picture, there are error marks under sigInfo which indicate 'Variable may not be initialized!'. This is really annoying during the development with mbeddr because it could happen often that we declare an instance of struct and initialize it somewhere else. It would be nice if we change the error severity to warning.

[image]

danielratiu, 2016-12-05:
Hi,

have you marked the parameter of createCanonicalRequest as OUT ?

[image: Inline image 1]

Dan

huibo-shi, 2016-12-07:
Hi Danielratiu,

After mark the parameter as OUT, the error is disappeared. Thanks.

closed 2016-12-07 by huibo-shi