mbeddr/mbeddr.core #1594: Interpreter optimization

ftomassetti, 2016-12-06:
It seems to me that `InterpreterEvaluationHelper` has a performance issue caused by the fact that this call is repeated for each evaluation:

```
IInterpreter interpreter = getInterpreter(getCategory());
```

By storing this value and reusing it across different evaluation performance seem to improve significantly for me.

I am not aware of all the consequences of doing that and I suppose that if someone changed the interpreter definition having it cached could be not a good idea.

markusvoelter, 2016-12-06:
We had found and fixed this bug a while ago, it is fixed on master. 

We fixed it through caching. This means that when developing interpreters, you want to switch that caching off. See the action in the Tools menu.

ftomassetti, 2016-12-07:
Maybe I did not explain well what I have noticed: to me it seems that the call `getInterpreter(getCategory());` is quite slow and by doing it once in the constructor instead of redoing it at every invokation of the interpreter there are significant performance gains. I have just duplicated `InterpreterEvaluationHelper` in the project and did the change there. I thought you could be interested in doing the change also in mbeddr.

bkolb, 2016-12-07:
We are caching exactly that, however, the cache is currently on the level of `IETS3ExprEvalHelper`. I will put that down into the core of the interpreter.

commit 61b1cc0f (2016-12-07): mpsutil: added caching for interpreters. Enabled by default. Option to disable in the tools menu for dev. purposes
fixed #1594
closed 2016-12-07 by bkolb