mbeddr/mbeddr.core #1595: Large consumption of memory in the de.itemis.mps.editor.celllayout package

ftomassetti, 2016-12-06:
I am running out of memory and it seems that a lot of memory is allocated by classes from this package (21 classes in the top 200 classes allocating more memory).

Below I report the data in CSV, for people involved in a certain project I can give you the complete list.

```csv
"Class Name - Live Objects";"Live Bytes [%]";"Live Bytes";"Live Objects"
"java.util.HashMap$Node";"13.07%";"170144352";"5317011"
"jetbrains.mps.util.Pair";"9.08%";"118259640";"4927485"
"jetbrains.mps.smodel.adapter.structure.concept.SConceptAdapterById";"8.63%";"112368624";"4682026"
"char[]";"7.36%";"95844768";"872910"
"java.lang.Object[]";"5.33%";"69380352";"1371218"
"java.util.HashMap$Node[]";"4.22%";"54938256";"239569"
"byte[]";"4.18%";"54471512";"117948"
"int[]";"2.87%";"37386496";"568600"
"de.itemis.mps.editor.celllayout.boxmodel.CalcOnWriteFrame";"2.69%";"35081400";"877035"
"de.itemis.mps.editor.celllayout.boxmodel.Size";"2.68%";"34882320";"1453430"
"de.itemis.mps.editor.celllayout.boxmodel.DefaultLayoutBox";"2.29%";"29805000";"745125"
"jetbrains.mps.nodeEditor.cells.TextLine";"2.22%";"28945392";"172294"
"java.util.TreeMap$Entry";"1.87%";"24363720";"609093"
"java.util.HashMap";"1.67%";"21758448";"453301"
"java.lang.String";"1.58%";"20628216";"859509"
"jetbrains.mps.smodel.SNode";"1.51%";"19670280";"351255"
"java.util.WeakHashMap$Entry";"1.04%";"13602320";"340058"
"java.util.LinkedHashMap$Entry";"1.00%";"13036240";"325906"
"jetbrains.mps.smodel.SNodeId$Regular";"1.00%";"13013448";"542227"
"de.itemis.mps.editor.celllayout.boxmodel.CalcOnWriteFrame$1";"0.95%";"12411824";"775739"
"java.util.TreeMap";"0.80%";"10426272";"217214"
"java.util.zip.ZipEntry";"0.72%";"9338320";"116729"
"jetbrains.mps.nodeEditor.cells.EditorCell_Collection";"0.69%";"9010344";"53633"
"java.util.ArrayList";"0.64%";"8360112";"348338"
"jetbrains.mps.util.ListMap$MyEntry";"0.61%";"7954392";"331433"
"gnu.trove.THashSet";"0.60%";"7830720";"195768"
"jetbrains.mps.smodel.InterfaceSNode";"0.55%";"7221824";"112841"
"de.itemis.mps.editor.celllayout.boxmodel.EditorCellBoxModel$1";"0.54%";"7016280";"175407"
"jetbrains.mps.editor.runtime.style.TopLevelStyleMap";"0.53%";"6902208";"287592"
"gnu.trove.TIntObjectHashMap";"0.53%";"6879504";"143323"
"long[]";"0.52%";"6732960";"5721"
"jetbrains.mps.nodeEditor.cells.EditorCell_Constant";"0.49%";"6414576";"47166"
"de.itemis.mps.editor.celllayout.boxmodel.EditorCellBoxModel";"0.45%";"5841760";"146044"
"de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter";"0.36%";"4750416";"53982"
"jetbrains.mps.editor.runtime.style.StyleImpl";"0.35%";"4601472";"143796"
"de.itemis.mps.editor.celllayout.runtime.LayoutableAdapter";"0.33%";"4278720";"89140"
"jetbrains.mps.smodel.StaticReference";"0.32%";"4200440";"105011"
"java.util.LinkedList$Node";"0.32%";"4152096";"173004"
"jetbrains.mps.nodeEditor.cells.EditorCell_Label$LocalHome";"0.31%";"3990600";"166275"
"jetbrains.mps.nodeEditor.cells.EditorCell_Label$LocalEnd";"0.31%";"3990600";"166275"
"jetbrains.mps.nodeEditor.cells.EditorCell_Label$MoveRight";"0.31%";"3990600";"166275"
"jetbrains.mps.nodeEditor.cells.EditorCell_Label$MoveLeft";"0.31%";"3990600";"166275"
"java.util.LinkedHashMap";"0.31%";"3973592";"70957"
"jetbrains.mps.util.ListMap$MyEntry[]";"0.29%";"3774456";"146601"
"de.itemis.mps.editor.celllayout.runtime.AbstractLayoutableAdapter$2";"0.26%";"3434928";"143122"
"de.itemis.mps.editor.celllayout.runtime.AbstractLayoutableAdapter$1";"0.26%";"3434928";"143122"
"jetbrains.mps.util.ListMap";"0.26%";"3433632";"143068"
"jetbrains.mps.internal.collections.runtime.ListSequence";"0.26%";"3358488";"139937"
"java.util.WeakHashMap$Entry[]";"0.24%";"3131136";"13824"
"java.util.HashSet";"0.24%";"3120464";"195029"
"java.lang.Class";"0.23%";"3057864";"27570"
"jetbrains.mps.nodeEditor.cells.EditorCell_Property";"0.23%";"2971728";"20637"
"jetbrains.mps.smodel.adapter.ids.SLanguageId";"0.21%";"2775328";"86729"
"com.intellij.openapi.vfs.impl.ArchiveHandler$EntryInfo";"0.20%";"2664360";"66609"
"java.util.ArrayList$Itr";"0.20%";"2576288";"80509"
"com.intellij.util.text.ByteArrayCharSequence";"0.20%";"2541240";"105885"
"jetbrains.mps.smodel.SNodePointer";"0.19%";"2496576";"104024"
"jetbrains.mps.smodel.SReference[]";"0.19%";"2487472";"103645"
"java.util.concurrent.ConcurrentHashMap$Node";"0.19%";"2485824";"77682"
"jetbrains.mps.internal.collections.runtime.MapSequence";"0.19%";"2412096";"150756"
"java.util.concurrent.ConcurrentLinkedQueue$Node";"0.18%";"2288496";"95354"
"java.util.LinkedList";"0.17%";"2224160";"69505"
"jetbrains.mps.smodel.adapter.ids.SConceptId";"0.16%";"2137824";"89076"
"java.util.TreeMap$EntrySet";"0.16%";"2119840";"132490"
[... 111 lines cut ...]
"jetbrains.mps.project.structure.modules.Dependency";"0.02%";"232104";"9671"
"jetbrains.mps.editor.runtime.impl.cellMenu.DefaultReferenceSubstituteInfoActionsFactory";"0.02%";"220800";"6900"
"de.itemis.mps.editor.celllayout.layout.IndentLayout";"0.02%";"218624";"13664"
"jetbrains.mps.internal.collections.runtime.impl.OfTypeSequence";"0.02%";"217728";"9072"
"java.util.Hashtable$Entry[]";"0.02%";"215272";"2075"
"jetbrains.mps.util.WeakSet";"0.02%";"214176";"13386"
"jetbrains.mps.smodel.SModel$ImportElement";"0.02%";"212400";"8850"
"java.util.concurrent.ConcurrentHashMap";"0.02%";"209856";"3279"
"java.lang.ref.SoftReference";"0.02%";"208880";"5222"
"java.util.LinkedList$ListItr";"0.02%";"208832";"6526"
"short[]";"0.02%";"205984";"292"
"jetbrains.mps.smodel.adapter.ids.SPropertyId";"0.02%";"205224";"8551"
"jetbrains.mps.nodeEditor.cellLayout.CellLayout_Vertical";"0.02%";"204928";"12808"
"jetbrains.mps.smodel.adapter.structure.language.SLanguageAdapterById";"0.02%";"200952";"8373"
"jetbrains.mps.smodel.adapter.structure.MetaAdapterFactory$LangKey";"0.02%";"200952";"8373"
"jetbrains.mps.smodel.adapter.ids.SReferenceLinkId";"0.02%";"198864";"8286"
"jetbrains.mps.smodel.adapter.structure.property.SPropertyAdapterById";"0.01%";"194232";"8093"
"jetbrains.mps.editor.runtime.cells.KeyMapImpl";"0.01%";"192608";"6019"
"de.slisson.mps.editor.multiline.cells.NewLineAction";"0.01%";"192608";"12038"
"jetbrains.mps.ide.platform.watching.FileSystemListenersContainer$Node";"0.01%";"191744";"5992"
"jetbrains.mps.openapi.editor.cells.KeyMap$ActionKey";"0.01%";"190248";"7927"
"jetbrains.mps.nodeEditor.cellLayout.CellLayout_Indent";"0.01%";"187904";"11744"
"java.util.concurrent.locks.ReentrantReadWriteLock$NonfairSync";"0.01%";"187776";"3912"
"java.beans.PropertyChangeListener[]";"0.01%";"186360";"7667"
```

slisson, 2016-12-07:
Memory consumption for the classes from "de.itemis.mps.editor.celllayout" is reduced to < 25%.

alexanderpann, 2024-07-14:
@slisson can we close it? There is no explanation for why this ticket was reopened.

alexanderpann, 2025-03-12:
Let's close the ticket because so much changed in the meantime. If there are still issues with memory consumption, a new ticket can be created.

closed 2025-03-12 by alexanderpann