mbeddr/mbeddr.core #1603: No refresh of PlantUML-Visualization after first SVG per MPS-Process

mgm-sp, 2016-12-15:
Hi,

[Just some info that may have something to do with the issue below: http://mbeddr.com/download.html#RequiredToolsAndVersions lists the Graphviz-Version 2.30 als being supported. However, when installing mbeddr from mbeddr-win-setup-298-MPS-3.4.2.exe version 2.38 ships.]

I have the problem, that after the first PlantUML-visualization (i use github/master with MPS 3.4.2) i do not get any new Visualization, i always see the first one that i did with the current MPS-Process (also holds for using build 298 from the releases). After restart i can render another one - but just one :( .
It doesn't work for mbeddr-tutorial nor for my own IVisualizable-Implementation. 
But for each visualization-call i get one more entry in SVGViewer's history (where i can step forward and backward) but always the same SVG is being displayed.

My System: 
- Windows 7
- MPS 3.4.2 
- GraphViz 2.38
- mbeddr github/master (built from source)

If you need more info, let me know.

Regards,
Mirko

mgm-sp, 2016-12-16:
Hi,

today i installed nightly-298-MPS-3.4.2 on a completely fresh system. The effect is the same: visualization seems not to render more than the first one (forward/backward do remember the selected categories but no re-draw is done).

Regards,
Mirko

mgm-sp, 2017-01-04:
Hi,

i debugged a little bit and realized that the red-marked code in the screenshot below is being entered only for the first start. Is there the possibility that _stopThenRun_ checks for a condition that never turns to be true in (my?) MPS/Batik, except for the first start, for some reason?

[image]

Regards,
Mirko

coolya, 2017-01-09:
It seems to be some classloading related problem.

I can reproduce the problem.

```
java.lang.Exception: loader constraint violation in interface itable initialization: when resolving method "org.apache.batik.dom.AbstractNode.addEventListener(Ljava/lang/String;Lorg/w3c/dom/events/EventListener;Z)V" the class loader (instance of jetbrains/mps/classloading/ModuleClassLoader) of the current class, org/apache/batik/dom/svg/SVGOMSVGElement, and the class loader (instance of <bootloader>) for interface org/w3c/dom/events/EventTarget have different Class objects for the type org/w3c/dom/events/EventListener used in the signature
	at com.mbeddr.mpsutil.plantuml.pluginSolution.plugin.PlantUMLSVGDocumentLoader.run(PlantUMLSVGDocumentLoader.java:90)
```

closed 2017-01-10 by coolya

mgm-sp, 2017-01-11:
It now renders correctly but the linking out of the svg seems to work no more. i cross-checked the MPS-generated String via http://www.plantuml.com/plantuml. There it works and correctly links into MPS.

commit 60e0a21a (2017-01-10): platform: fix rendering problem for plantuml

plantuml was only rendering once per MPS instance because the update manager was'to
canceled correctly in the custom complete handler.

fix #1603