mbeddr/mbeddr.core #1604: Collapse/Expand does not work properly for editors which use "CellModel_Diagram"

Nandha1712, 2017-01-09:
I have created a School language which contains Teachers and ClassRooms. I created diagram editor for School and its contents using "CellModel_Diagram". Changing to the diagram view using push editor hints, toggles the view to diagram view. Now if we try to collapse, the School using Ctrl+- or corresponding button, the graphical canvas stays. The view is getting more complicated, when there are multiple Schools.

[image] 
[image] 
[image]

slisson, 2017-01-10:
It's an MPS bug: https://youtrack.jetbrains.com/issue/MPS-25164

Nandha1712, 2017-07-20:
This is fixed in MPS 2017.1.2

closed 2017-07-20 by Nandha1712