mbeddr/mbeddr.core #1605: Dropping from diagram palette does not work properly

Nandha1712, 2017-01-10:
1. I have created a language which uses, mbeddr's CellModel_Diagram.
2. When i tried to drop a component from diagram palette, the dropping action is not happening properly.
3. The canDrop() section gets executed. But drop() section does not get executed though canDrop() returns true explicitly.
[image]

slisson, 2017-01-13:
This feature only works with palette entries of type AbstractDragPaletteEntry. This is also known to not work on windows.

Nandha1712, 2017-01-18:
Thank you. But may i know why drop() does not work on windows? Is it due to the library jgraphx? Since it is built using java, it should work on all the Operating Systems right?

slisson, 2017-01-18:
The only thing I know is that on windows the drop event is not created. I already tried to find the reason but no luck so far.

Nandha1712, 2017-01-18:
Okay. Thank you

Nandha1712, 2017-05-15:
@slisson Is there any progress regarding this issue?

slisson, 2017-05-15:
No, at the moment we don't need this feature and don't have time to work on this.

Nandha1712, 2017-05-15:
Okay .

slisson, 2017-05-18:
Fixed in f32a8896762c8ca383b2666258a4b1220c335a3f

closed 2017-05-18 by slisson