mbeddr/mbeddr.core #1620: Generalize path variable support in build confiugrations and add support for specifying simple variables and environment variables

fbacha, 2017-02-02:
- Prefix path variables as the following: ${path_var:smartmeter.code}
- Prefix environment  variables as the following: ${env_var:USERPROFILE}
- Propose custom variables: ${HOME}
All these variables should be evaluated and substitued by mbeddr

coolya, 2017-02-07:
Path variables know to MPS are already supported using `${smartmeter.code.home}`. At least in all the path related text fields of the build config.

fbacha, 2017-02-09:
We would like to generalize using variables to define paths in build config. In addition to MPS path variables, we will give the possibility also to use environment variables when defining paths. Of course,  those variables values will be dynamically interpreted and integrated in the generated makefile.

alexanderpann, 2022-04-27:
Closing because it seems to be fixed in https://github.com/mbeddr/mbeddr.core/pull/1647.

closed 2022-04-27 by alexanderpann