mbeddr/mbeddr.core #1641: Gradle build does not copy mbeddr plug-ins to MPS plugins folder

stephaneberle9, 2017-03-06:
When running the Gradle build (i.e., ./gradlew) for mbeddr with a fresh MPS installation under the hoods, the following plug-ins are missing in the MPS installation's plugins folder after the build has finished:

de.itemis.mps.nativelibs.loader
com.mbeddr.mpsutil.actionsfilter
mbeddr.spawner

As a consequence, it is impossible for the user to use mbeddr.

This problem has been observed under Linux and Windows. Permissions were sufficient in both cases, i.e., this seems not to be the reason for the problem.

coolya, 2017-03-06:
Did you execute the install target?

```
gradlew.bat install "-PMPS_Installation=<path to MPS>"
```

If so, when copying fails the build will fail and give you an error message. Or was it silently failing?

stephaneberle9, 2017-03-06:
It does fails silently when invoking the build without any specific target. When invoking the build as mentioned above, I get the following error:

16:47:21.706 [DEBUG] [org.gradle.internal.operations.DefaultBuildOperationWorkerRegistry] Worker root.5 started (1 in use).
16:47:21.706 [INFO] [org.gradle.execution.taskgraph.AbstractTaskPlanExecutor] :build:com.mbeddr:languages:install_spawner (Thread[Daemon worker Thread 3,5,main]) started.
16:47:21.706 [LIFECYCLE] [class org.gradle.internal.buildevents.TaskExecutionLogger] :build:com.mbeddr:languages:install_spawner
16:47:21.706 [DEBUG] [org.gradle.api.internal.tasks.execution.ExecuteAtMostOnceTaskExecuter] Starting to execute task ':build:com.mbeddr:languages:install_spawner'
16:47:21.708 [DEBUG] [org.gradle.api.internal.tasks.execution.SkipUpToDateTaskExecuter] Determining if task ':build:com.mbeddr:languages:install_spawner' is up-to-date
16:47:21.806 [INFO] [org.gradle.api.internal.tasks.execution.SkipUpToDateTaskExecuter] Executing task ':build:com.mbeddr:languages:install_spawner' (up-to-date check took 0.098 secs) due to:
  No history is available.
16:47:21.806 [DEBUG] [org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter] Executing actions for task ':build:com.mbeddr:languages:install_spawner'.
16:47:21.809 [DEBUG] [org.gradle.api.internal.tasks.execution.ExecuteAtMostOnceTaskExecuter] Finished executing task ':build:com.mbeddr:languages:install_spawner'
16:47:21.809 [LIFECYCLE] [class org.gradle.internal.buildevents.TaskExecutionLogger] :build:com.mbeddr:languages:install_spawner FAILED
16:47:21.809 [INFO] [org.gradle.execution.taskgraph.AbstractTaskPlanExecutor] :build:com.mbeddr:languages:install_spawner (Thread[Daemon worker Thread 3,5,main]) completed. Took 0.103 secs.
16:47:21.809 [DEBUG] [org.gradle.internal.operations.DefaultBuildOperationWorkerRegistry] Worker root.5 completed (0 in use)
16:47:21.809 [DEBUG] [org.gradle.execution.taskgraph.AbstractTaskPlanExecutor] Task worker [Thread[Daemon worker Thread 3,5,main]] finished, busy: 2 mins 44.882 secs, idle: 0.003 secs
16:47:21.810 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]
16:47:21.810 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] FAILURE: Build failed with an exception.
16:47:21.810 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]
16:47:21.810 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] * What went wrong:
16:47:21.810 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] Execution failed for task ':build:com.mbeddr:languages:install_spawner'.
16:47:21.811 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] > Could not copy file 'W:\github\mbeddr.core\artifacts\spawner\mbeddr.spawner\languages\Eclipse.Debugger-src.jar' to 'C:\Program Files\JetBrains\MPS 3.4\plugins\mbeddr.spawner\languages\Eclipse.Debugger-src.jar'.
16:47:21.811 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter]
16:47:21.811 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] * Try:
16:47:21.811 [ERROR] [org.gradle.internal.buildevents.BuildExceptionReporter] Run with --stacktrace option to get the stack trace.
16:47:21.811 [LIFECYCLE] [org.gradle.internal.buildevents.BuildResultLogger]
16:47:21.811 [LIFECYCLE] [org.gradle.internal.buildevents.BuildResultLogger] BUILD FAILED
16:47:21.811 [LIFECYCLE] [org.gradle.internal.buildevents.BuildResultLogger]
16:47:21.811 [LIFECYCLE] [org.gradle.internal.buildevents.BuildResultLogger] Total time: 2 mins 48.304 secs
16:47:21.812 [DEBUG] [org.gradle.cache.internal.btree.BTreePersistentIndexedCache] Closing cache fileHashes.bin (W:\github\mbeddr.core\.gradle\3.1\taskArtifacts\fileHashes.bin)
16:47:21.815 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasing lock on task history cache (W:\github\mbeddr.core\.gradle\3.1\taskArtifacts).
16:47:21.823 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.resolveengine.store.CachedStoreFactory] Resolution result cache closed. Cache reads: 6, disk reads: 2 (avg: 0.0 secs, total: 0.001 secs)
16:47:21.823 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.resolveengine.store.CachedStoreFactory] Resolution result cache closed. Cache reads: 0, disk reads: 0 (avg: 0.0 secs, total: 0.0 secs)
16:47:21.823 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.resolveengine.store.ResolutionResultsStoreFactory] Deleted 2 resolution results binary files in 0.001 secs
16:47:21.824 [DEBUG] [org.gradle.cache.internal.btree.BTreePersistentIndexedCache] Closing cache artifact-at-repository.bin (C:\Users\steph\.gradle\caches\modules-2\metadata-2.21\artifact-at-repository.bin)
16:47:21.824 [DEBUG] [org.gradle.cache.internal.btree.BTreePersistentIndexedCache] Closing cache module-artifacts.bin (C:\Users\steph\.gradle\caches\modules-2\metadata-2.21\module-artifacts.bin)
16:47:21.824 [DEBUG] [org.gradle.cache.internal.btree.BTreePersistentIndexedCache] Closing cache module-metadata.bin (C:\Users\steph\.gradle\caches\modules-2\metadata-2.21\module-metadata.bin)
16:47:21.824 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasing lock on artifact cache (C:\Users\steph\.gradle\caches\modules-2).
16:47:21.827 [DEBUG] [org.gradle.cache.internal.btree.BTreePersistentIndexedCache] Closing cache plugin-use-metadata.bin (C:\Users\steph\.gradle\caches\3.1\plugin-resolution\plugin-use-metadata.bin)
[... cut ...] C:\Users\steph\.gradle\daemon,pid=12956,idleTimeout=10800000,daemonOpts=-XX:MaxPermSize=256m,-XX:+HeapDumpOnOutOfMemoryError,-Xmx1024m,-Dfile.encoding=windows-1252,-Duser.country=FR,-Duser.language=fr,-Duser.variant]} (build should be done).
16:47:22.163 [DEBUG] [org.gradle.launcher.daemon.client.DaemonClientConnection] thread 1: dispatching class org.gradle.launcher.daemon.protocol.Finished
16:47:22.164 [DEBUG] [org.gradle.launcher.daemon.client.DaemonClientConnection] thread 1: connection stop

coolya, 2017-03-06:
I still suspect this is a permission problem. I think to work around this we should start deploying our plugins into the users plugins location (in the user folder .MPSXX) rather than in the real installation. 

We have be discussing this for quite some time and should finally implement this. The user will always have permissions to this folder and it will persist even MPS reinstallations at that location.

@DomenikP what do you think?

stephaneberle9, 2017-03-20:
What is strange about this it that the old Ant-based build WAS able to copy the necessary artifacts to the MPS plugins folder. More precisely, in case that MPS was installed under C:\Program Files (x86)\JetBrains, the Ant-based build did not copy the artifacts, when it was invoked normally, but it was pretty well able to do so, when it was run from a console that has been opened using the "Execute as Administrator" option.

My proposal would be to investigate why the new Gradle-based build behaves differently in that particular aspect, i.e., why it is unable to inherit/run in an elevated context when being started from a console with administrator privileges.

coolya, 2017-03-20:
This is caused because gradle by default uses daemons to execute your build. This means the gradle instance from you console only sends a command to build a given project to that daemon. This daemon runs as the current user.

Can you [disable](https://docs.gradle.org/current/userguide/gradle_daemon.html#sec:disabling_the_daemon) the daemon to verify this?

alexanderpann, 2024-11-25:
So much changed since this issue was created, it is therefore obsolete.

closed 2024-11-25 by alexanderpann