mbeddr/mbeddr.core #1642: Unnecessary dependency on Git in gradlew.bat

stephaneberle9, 2017-03-06:
When running the Gradle build for mbeddr using the gradlew.bat script locally under Windows, it almost immediately fails. This seems to be caused by the missing entry for Git on the PATH environment variable (the problem actually goes away as soon as Git is added to the PATH). However, it would be more adequate if the Gradle build wouldn't require Git at all when being executed locally.

coolya, 2017-03-06:
Can you please post the exception so that I can trace down where the dependency is, because its most likely a bug and shouldn't be there.

stephaneberle9, 2017-03-06:
Sure, here you go:

Starting a Gradle Daemon (subsequent builds will be faster)
:buildSrc:compileJava UP-TO-DATE
:buildSrc:compileGroovy UP-TO-DATE
:buildSrc:processResources UP-TO-DATE
:buildSrc:classes UP-TO-DATE
:buildSrc:jar UP-TO-DATE
:buildSrc:assemble UP-TO-DATE
:buildSrc:compileTestJava UP-TO-DATE
:buildSrc:compileTestGroovy UP-TO-DATE
:buildSrc:processTestResources UP-TO-DATE
:buildSrc:testClasses UP-TO-DATE
:buildSrc:test UP-TO-DATE
:buildSrc:check UP-TO-DATE
:buildSrc:build UP-TO-DATE
Local build detected. Build will be a SNAPSHOT.

FAILURE: Build failed with an exception.

* Where:
Build file 'W:\github\mbeddr.core\build\com.mbeddr\build.gradle' line: 72

* What went wrong:
A problem occurred evaluating project ':build:com.mbeddr'.
> String index out of range: 7

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

commit c73f4257 (2017-03-06): build: remove git dependency for the build. fix #1642
closed 2017-03-06 by coolya