mbeddr/mbeddr.core #1646: Cannot use a predefined variable in For loop

aswinsugumaran, 2017-03-08:
Hello, 
       I have a variable declared outside the for loop.When i try to reuse the same variable again in initialization part of the for loop its not visible. I have tried this some other loops statements which works fine. Only for the For loop it doesn't seem to work.Any suggestion would be of great help.

[image]

aswinsugumaran, 2017-03-09:
a

aswinsugumaran, 2017-03-09:
@coolya  from what i inferred we have this constraint named ForVarRef_Constraints which returns false cause of iterator is null at time of for loop initialization. I set the constraint to be true manually which works now. Is this the proper solution for this or any other alternate solution is there?

coolya, 2017-03-09:
I will have a look at this tomorrow. From you error description I don't really understand what the issue is. I guess `ix` is not in scope, right?

Can give a text example what you want code you would like to write?

aswinsugumaran, 2017-03-09:
@coolya  the ix is in the scope. 
Steps to reproduce 
1. Create a local variable
2. Try to use the local variable in the simple for loop
3. The local variable will not be available in the code completion menu of for loop initialization part.

[image]

coolya, 2017-03-09:
What you do at that location is defining the iterator. There is no way to use `ix` at that location. 

Once again, please give me a **text**  example of the code you would like to be able to enter.

aswinsugumaran, 2017-03-09:
Example
int a;

for( <the variable 'a' should be listed here when i use the code completion menu > ;    ; )

coolya, 2017-03-09:
I don't think this makes much sense. According to the C99 spec 6.8.5.3 a for statements is defined like this:

```for ( clause-1 ; expression-2 ; expression-3 ) statement```

Where clause-1 can either be a declaration, which would make it a variable with a scope within the for loop, or a void expression.

The void expression case is not supported by mbeddr, you are only allowed to declare a variable at that location. 

What should work is something like 
```

int16 a;

for (int16 i = a; a++ ;) {
}

```

DineshLMS, 2017-03-09:
@coolya I don't think it was implemented keeping C99 standard in mind. It was working previously. I couldn't understand how it was working in previous releases (25/11/2016) and not working all of sudden after migration to MPS 3.4.3 and gradle build. any idea ?

slisson, 2017-03-09:
It works if I delete the constraint ForVarRef.canBeChild.

DineshLMS, 2017-03-09:
@slisson That is what @aswinsugumaran is also proposing now and asking @coolya to give suggestions or comments

aswinsugumaran, 2017-03-31:
https://github.com/mbeddr/mbeddr.core/pull/1661

closed 2017-03-31 by aswinsugumaran