mbeddr/mbeddr.core #1651: math.parentheses does not consider Darcula style

hnehls, 2017-03-10:
When using Darcula-Theme, the parentheses, rendered by math.parentheses, stay black but should be white. Also I could not find any corresponding style attribute (just 'bracket-color' and 'background-color' but both do not match, I guess and anyway, they have no effect).
Same issue with DivisionEditor.

In math.sum it works fine.

[image]

hnehls, 2017-03-10:
> In math.sum it works fine

Well, for the brackets, but not for the sum symbol ;)

slisson, 2017-03-10:
There should be "symbol-color" style. Maybe that one works. But I will fix the default color.

commit 0524c94e (2017-03-10): Math: Changed default symbol color to the text color

Fixes #1651
closed 2017-03-10 by slisson