mbeddr/mbeddr.core #1659: Two buildconfigurations in one model give unintuitive error

eugenschindler, 2017-03-23:
When I add two build configurations to a model, I get an error for both build configurations that looks like this:
[image]
Probably this is due to the fact that no two build configurations are allowed per model. But then I would expect an error like: "only 1 build configuration per model allowed".

coolya, 2017-04-19:
How did you mange to create a second one?

Other than programatically adding one this shouldn't be possible?

eugenschindler, 2017-04-19:
I just right-click on a model that uses the com.mbeddr.core.buildconfig language, then select New --> c.m.core.buildconfig --> BuildConfiguration. I can do this 2 times and then the BuildConfiguration root node disappears from the "New" submenu. I am using mbeddr nightly 380.