mbeddr/mbeddr.core #1663: Filling in function arguments is quite un-explorable

eugenschindler, 2017-03-27:
Currently, if I have to enter a C function, I have to go back to the function definition in order to know which arguments I can fill in. Since this is a case that is often used, maybe some mechanism can be added to easily add some example parameters to a function call? Maybe an intention?

markusvoelter, 2017-03-27:
Tried pressing Ctrl-P?

eugenschindler, 2017-03-28:
Yeah, that helps :) Thanks for the hint, Markus!
Would it be possible to do it like you see in "normal" IDEs, that you see which parameter to type, as you type them?

slisson, 2017-03-28:
It's implemented in MPS and has to be changed there.

closed 2017-03-28 by slisson

eugenschindler, 2017-04-12:
Thanks for the information, @slisson. Do you know if there is already a YouTrack issue for it that I can +1 or did no-one make one yet?

slisson, 2017-04-12:
I don't know of any existing issue.

eugenschindler, 2017-04-26:
I made an issue here: https://youtrack.jetbrains.com/issue/MPS-25853