mbeddr/mbeddr.core #1680: Two unnamed context actions tabs showing

ftomassetti, 2017-04-18:
[image]

Nothing is showing when I select the first tab, while on the second I can see all the context actions I have added.

Not sure why this is happening. Any idea?

aswinsugumaran, 2017-05-11:
I faced this problem too. As of temporarily i manually rebuild the particular plugin responsible for the Context-actions and the tabs are gone!It works as expected.

ftomassetti, 2017-05-12:
Hi @aswinsugumaran we have rebuilt it many times but we have still the double tabs unfortunately

ftomassetti, 2017-05-12:
I suspect there are two instances of `ToolComponent` but I do not why or how to prevent this from happening. Both the instances have the search widget, but just one list the context actions. Could be possible that somehow there are two versions of the mpsutil plugin?

ftomassetti, 2017-06-29:
I understand this is a very minor issue and there are hundreds of more pressing issues w.r.t. this one but I was wondering if there was any progress on this one

sergej-koscejev, 2017-06-29:
The cause is that there's a Context Actions tool in MPS now, more or less copied from the mbeddr tool (but using the transformation menu language to define actions). Since they both have the same name, they show up in the same place but as two tabs. The easiest fix would be to rename the mbeddr tool to "Context Actions (mbeddr)".

markusvoelter, 2017-06-29:
Are you sure that this is the issue? I am pretty sure we had the double-tab issue before MPS had context actions.

closed 2017-06-29 by slisson

aswinsugumaran, 2017-07-25:
Now i have two context-actions are on my window after this fix. Any idea to replace the context-actions from MPS ?

sergej-koscejev, 2017-07-25:
@aswinsugumaran disable "MPS Context Actions Tool" plugin

aswinsugumaran, 2017-07-25:
@sergej-koscejev  works thanks :)

aswinsugumaran, 2017-07-25:
@sergej-koscejev  is there a chance we can enable this setting defaultly off from Mbeddr context actions ?

sergej-koscejev, 2017-07-25:
I don't fully understand what you mean, please elaborate.

aswinsugumaran, 2017-07-25:
Instead of going to settings and disabling the MPS context actions, can we do this inside Mbeddr context actions so that it automatically removes the MPS context actions?

sergej-koscejev, 2017-07-25:
I don't know if it's possible but I wouldn't do it. You can remove the MPS context actions plugin when you package your own MPS-based RCP.

vsnarendran, 2017-07-27:
Yes Aswin. We can remove the plugin MPS201712\plugins\context-tool that will remove the tab from mps.

commit 368520d5 (2017-06-29): ContextActions: Renamed tool to "Context Actions (mbeddr)"

Fixes #1680