mbeddr/mbeddr.core #1682: Explicit initialization of structures with boolean values using a default init "{0}" leads to type-system error

danielratiu, 2017-04-23:
struct s1 {
 boolean b1;
 boolean b2:
}

s1 v = {0} <-- type error (see screenshot)

[image]

andreasgrosche, 2017-04-24:
Regarding pure C, the initializer list initializes the named elements of the struct, starting with the first element.
The types of the elements in the initializer list have to match the types of the struct members in the same order.
The missing elements are initialized "implicitly the same as objects that have static storage duration" ("false" in this case).

E.g. the following code works for C (the first element is explicitely initialized with false and all other elements are implicitly initialized with the default value for static objects "false"):

```
struct s1 { 
  boolean b1; 
  boolean b2; 
};
                                             
s1 v = {false};  
```

Unfortunately, mbeddr complains, "wrong number of arguments".

For initialization in general, see also [http://stackoverflow.com/questions/11152160/initializing-a-struct-to-0](http://stackoverflow.com/questions/11152160/initializing-a-struct-to-0).

danielratiu, 2017-04-24:
@andreasgrosche 
AFAIK each structure can be initialized (default) with "structVar = {0};"

We should allow this in mbeddr as well.

coolya, 2017-04-24:
@danielratiu an empty initialiser list should work and give you the expected result.

coolya, 2017-04-24:
@andreasgrosche yes C permits that but mbeddr currently not. But it would make good first contribution if you like 😉  We are always happy to see pull requests.