mbeddr/mbeddr.core #1687: GrammarCells wrap do not consider constraints

ftomassetti, 2017-04-27:
I have a case in which I have an editor containing a wrap grammar cell. The concept of the editor is not allowed in a certain context but the substitution happens anyway, resulting in an invalid model.

slisson, 2017-04-27:
What type of constraint is it? A wrapper substitute actions can only check the "canBeChild" constraint of the parent.

ftomassetti, 2017-04-27:
It is a canBeChild that applies to the new node being created to replace the existing one

slisson, 2017-04-27:
It's not possible to check the constraint before the node is created. As an alternative you can try to specify a condition in the inspector of the wrapper cell.

ftomassetti, 2017-05-01:
That does not seem to work. Consider this:

[image]

None of this is invoked.

In this case I have a number, when I type "in" two things happen: first this transformation is executed, wrapping the number in this concept, then a second transformation happens so that the new node (result of the first transformation) is inserted in a larger expression.

slisson, 2017-05-01:
I tried to reproduce it here: https://github.com/slisson/mbeddr-1687 (I used MPS 2017.1), but without success. In my case the condition is evaluated. Can you check the example a try to find the difference?

ftomassetti, 2017-05-03:
We are on MPS3.4.3 (we cannot migrate for the time being). I think it is relevant that this issue appear only when we have two transformations concatenate. The single wrapping action is never being executed alone, only when we type "in", triggering another transformation.

ftomassetti, 2017-05-03:
I have looked at your example and it seems to use only a wrapping element.

In my case I have a concept named `EqualToContextualExpression`. Its editor wraps an `Expression`.

When I edit code in a context in which `EqualToContextualExpression` can be used, it is replaced corrently. The print statements I have added are executed, so the conditions are evaluated.

However, in a context in which `EqualToContextualExpression` cannot be used, when I type `in` an `InRangeExpression` is added. It has this editor:

[image]

Now, when typing `in` I end up with my an `InRangeExpression` wrapped into a `EqualToContextualExpression`. In this case the conditions are not evaluated (no print statements). I think I have seen the `EqualToContextualExpression` used to wrap only the `value` element of the whole `InRangeExpression` other times but this is the case I report in the screenshot below:

[image]

I hope it helps figuring out why conditions are not evaluated

slisson, 2017-05-03:
A minimal example project would be even more useful. Then I don't have to (mis-)understand your description.

ftomassetti, 2017-05-03:
I have created one in https://github.com/ftomassetti/mbeddr1687b 

Also in this case I added some print and verified that they are not executed.

In the example solution I create `ExpressionsList`, type a number (e.g. 1) and type `in`. The whole expression is wrapped into an EqualToContextualExpression but the condition are not verified.

slisson, 2017-05-03:
This seems to be a bug in parser of the grammarCells. You can work around it by adding some constants to the grammar of EqualToContextualExpression. Then the concept is not added when it wasn't there before.

[image]

ftomassetti, 2017-05-15:
The workaround you suggested worked on our particular project. Thanks!

closed 2017-05-15 by slisson