mbeddr/mbeddr.core #1688: No such method found EditorCell_MathEnd.getTextOnEvent

ecost4, 2017-04-28:
Hi,

I'm trying to update my language to mps version 2017.1 and I'm getting the following error. This error occurred in math plugin in several operations. For example when we have ln(10) and try sum an integer value (ln(10) +1) the error occurred. If we sum two integer values (1 + 1) the error don't occur. The error 

You are expecting an method in mps class (EditorCell_MathEnd) that does not exist. This method was removed in:
Author:  Alexander Shatalin <Alexander.Shatalin@jetbrains.com>
Author date: 2 months ago (10/03/2017 11:43:27)
Commit date: 2 months ago (10/03/2017 11:43:58)
Commit hash: 80a813a612f6d7488781093f6a77490a8369a00c
Parent(s): 9ab6638bbc

MPS-25473 type-and-hold leads to freeze on mac

Contained in branches: upstream/master, origin/smmribeiro.001.optimizations, origin/master, origin/HEAD, smmribeiro.001.optimizations, master
Contained in tags: 2017.1.0, 2017.1

Can you help us?

**Stacktrace**:
  ERROR - #com.intellij.ide.plugins.PluginManager - java.lang.NoSuchMethodException: jetbrains.mps.nodeEditor.cells.EditorCell_Label.getTextOnEvent(java.awt.event.KeyEvent) 
java.lang.RuntimeException: java.lang.NoSuchMethodException: jetbrains.mps.nodeEditor.cells.EditorCell_Label.getTextOnEvent(java.awt.event.KeyEvent)
	at de.itemis.mps.editor.math.runtime.EditorCell_MathEnd.getTextOnEvent(EditorCell_MathEnd.java:97)
	at de.itemis.mps.editor.math.runtime.EditorCell_MathEnd.access$000(EditorCell_MathEnd.java:21)
	at de.itemis.mps.editor.math.runtime.EditorCell_MathEnd$1.run(EditorCell_MathEnd.java:70)
	at jetbrains.mps.smodel.WorkbenchModelAccess$2.run(WorkbenchModelAccess.java:108)
	at com.intellij.openapi.application.impl.ApplicationImpl.runReadAction(ApplicationImpl.java:926)
	at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:103)
	at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:53)
	at de.itemis.mps.editor.math.runtime.EditorCell_MathEnd.doProcessKeyTyped(EditorCell_MathEnd.java:68)
	at jetbrains.mps.nodeEditor.cells.EditorCell_Basic.processKeyTyped(EditorCell_Basic.java:453)
	at jetbrains.mps.nodeEditor.EditorComponentKeyboardHandler.processKeyTyped(EditorComponentKeyboardHandler.java:72)
	at jetbrains.mps.nodeEditor.EditorComponent.processKeyTyped(EditorComponent.java:2427)
	at jetbrains.mps.nodeEditor.EditorComponent$11.keyTyped(EditorComponent.java:594)
	at java.awt.AWTEventMulticaster.keyTyped(AWTEventMulticaster.java:240)
	at java.awt.Component.processKeyEvent(Component.java:6496)
	at javax.swing.JComponent.processKeyEvent(JComponent.java:2832)
	at java.awt.Component.processEvent(Component.java:6318)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4897)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
	at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:806)
	at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1074)
	at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:945)
	at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:771)
	at com.intellij.ide.IdeKeyboardFocusManager.dispatchEvent(IdeKeyboardFocusManager.java:27)
	at java.awt.Component.dispatchEventImpl(Component.java:4768)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4719)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:737)
	at java.awt.EventQueue$4.run(EventQueue.java:735)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:827)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:634)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:365)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.NoSuchMethodException: jetbrains.mps.nodeEditor.cells.EditorCell_Label.getTextOnEvent(java.awt.event.KeyEvent)
	at java.lang.Class.getDeclaredMethod(Class.java:2130)
	at de.itemis.mps.editor.math.runtime.EditorCell_MathEnd.getTextOnEvent(EditorCell_MathEnd.java:89)
	... 50 more

slisson, 2017-04-28:
NoSuchMethodException can only occur if when a class was compiled against a different version than the one loaded at runtime. This means the mbeddr version you are using was not built with MPS 2017.1. At the moment there is no mbeddr release available for MPS 2017.1.

ecost4, 2017-05-01:
I've downloaded from master branch. When have you expect a available release for 2017.1?

Eduardo Costa

ecost4, 2017-05-03:
Sorry Sascha, but it is important to us to know when we will have a release
available for MPS 2017.1.

We are updating our application, and if the mbdder won't have a release
soon, we can't proceed with this update.

Best regards

coolya, 2017-05-03:
As the method is called with reflection I think this a bug in the math editor code.

```
private string getTextOnEvent(KeyEvent keyEvent) { 
  try { 
    Method m = EditorCell_Label.class.getDeclaredMethod("getTextOnEvent", KeyEvent.class); 
    m.setAccessible(true); 
    return (string) m.invoke(this, keyEvent); 
  } catch (InvocationTargetException e) { 
    throw new RuntimeException(e); 
  } catch (IllegalAccessException e) { 
    throw new RuntimeException(e); 
  } catch (NoSuchMethodException e) { 
    throw new RuntimeException(e); 
  } 
}
```

slisson, 2017-05-03:
I didn't know reflection was involved here. In that case the version from master is fine. I fixed it there.

commit 15f9c74d (2017-05-03): Fixed #1688 No such method found EditorCell_MathEnd.getTextOnEvent
closed 2017-05-03 by slisson