mbeddr/mbeddr.core #1708: Auto-Layout not works properly in State machine graphical

aswinsugumaran, 2017-06-21:
[image]
Hi,

I have attached a sample model of state machine represented graphically,from the model in the composite state **State1** when i do auto layout, nothing happens and also there is transition appearing behind **state2** composite state.  Any suggestion to fix it would be of great help.

DomenikP, 2017-08-08:
@slisson 
Do you have an idea?

slisson, 2017-08-08:
Auto layout of edge across (sub-)diagram is not supported at the moment. Each (sub-)diagram is layouted individually.

aswinsugumaran, 2017-08-11:
@slisson  will there any update on this in upcoming days?

slisson, 2017-08-11:
The are no plans to implement this feature.

closed 2017-08-11 by aswinsugumaran