mbeddr/mbeddr.core #1711: The archive mps-sl-all.zip doesn't exist

tgunr, 2017-06-22:
Fresh clone of commit 22db46 onto Mac at /Sources/Github/mbeddr.core. 
Setup code/languages/build.properties with:
`# MPS installation
mps.home=/Applications/JetBrains/apps/MPS/ch-0/171.1342/MPS 2017.1.app/Contents
# Folder where MPS stores it's cache
mps.platform.caches=/Users/davec/Library/Caches/MPS2017.1
# Points to the root folder of the mbeddr.core repository
mbeddr.github.core.home=/Sources/GitHub/mbeddr.core
`

`cd /Sources/Github/mbeddr.core/code/languages/
./buildlanguages.sh
....
      [jar] Building jar: /Sources/GitHub/mbeddr.core/artifacts/mps-sl-all/de.itemis.mps.celllayout/languages/de.itemis.mps.celllayout/de.itemis.mps.editor.celllayout-generator.jar
[copyModels] Copying 1 file to /Sources/GitHub/mbeddr.core/tmp/mps-sl-all/customProcessors/copyModels/languages-de.itemis.mps.editor.celllayout-generator-template
[copyModels] Copying 5 files to /Sources/GitHub/mbeddr.core/tmp/mps-sl-all/customProcessors/copyModels/languages-de.itemis.mps.editor.celllayout-models
      [jar] Building jar: /Sources/GitHub/mbeddr.core/artifacts/mps-sl-all/de.itemis.mps.celllayout/languages/de.itemis.mps.celllayout/de.itemis.mps.editor.celllayout-src.jar

BUILD FAILED
/Sources/GitHub/mbeddr.core/code/languages/build.xml:126: The following error occurred while executing this line:
/Sources/GitHub/mbeddr.core/code/languages/build.xml:274: The following error occurred while executing this line:
/Sources/GitHub/mbeddr.core/code/languages/build.xml:221: The archive mps-sl-all.zip doesn't exist
`

coolya, 2017-07-19:
It seems like you have invoked the ant script directly. Would you mind please trying to build with gradle? 

There is a script called `gradlew` in the root of the repo. You can execute it without any parameters and it will build mbeddr.

Please also have a look at our wiki for detailed instructions:

https://github.com/mbeddr/mbeddr.core/wiki/Building-mbeddr

alexanderpann, 2022-04-27:
Closing because the issue was resolved.

closed 2022-04-27 by alexanderpann