mbeddr/mbeddr.core #1712: MPS version and nightly disappeared from file names

eugenschindler, 2017-06-27:
In https://github.com/mbeddr/mbeddr.core/releases, the files belonging to the release (like com.mbeddr.allInOne.zip) file used to have a postfix of the nightly build number and the MPS version (e.g. com.mbeddr.allInOne-410-MPS-3.4.3.zip). In the newest nightly (507), these numbers have disappeared. Would it be possible to make it so that these numbers are generated again? They are quite useful and important for configuration management.

eugenschindler, 2018-01-27:
Would it be possible to re-add this functionality? It would really help with packaging custom MPS with mbeddr and other packages for tutorial purposes (like lab sessions for students) in which I want to use the mbeddr plugins from github together with other MPS plugins.

coolya, 2018-01-29:
The problem is theses zips are packaged by MPS and by default the MPS scripts have no idea about the MPS version in use. Only the gradle scripts know which MPS version is used. I know it sounds strange. 

We removed the MPS from then when we moved publishing to github over from ant/bash to gradle to simplify the workflow. There are some places in our build pipeline that need to know the path of the zip file and they all would need to get aware of the MPS version. This is really not tiny refactoring as it might seem on the first place. I will have to investigate what breaks if we rename the zip. Maybe we can use some workaround the renames the zip just for uploading or something.

eugenschindler, 2018-01-31:
Hi Kolja, thanks for your quick reaction! I understand the problem. Probably this is a very simplistic view and I can imagine you have thought about it already, but regarding keeping the MPS version, maybe it helps to just have a file or some entry in the gradle properties file that knows the MPS version?
Regarding the "upload-name": personally, I think renaming (or copying, or renaming and then renaming back, if you still need the original filename) to a new file name and then uploading that file is not a workaround, it's quite a good mechanism to upload to github releases without touching the rest of the process. I know it's a million times more simple than mbeddr, but the plaintextgen travis-script does something like that (see https://github.com/DSLFoundry/mps-plaintextgen/blob/master/.travis.yml). The only thing with copying is: it may increase build times (don't know if that is a big issue on build servers).

wsafonov, 2020-01-10:
After reactivating releases on Github, versions of the published artifacts seem to be ok. Therefore closing this one.

closed 2020-01-10 by wsafonov