mbeddr/mbeddr.core #1736: Tutorials first time load - undefined variables bug

DorelComan, 2017-08-14:
I installed mbeddr with the windows setup and I loaded the tutorials. I deleted the cache before. Opening the tutorials I get this errors:

[image]

[image]

This variable is used here in the Project Libraries

[image]

eugenschindler, 2017-08-14:
Why is this an issue? It's just MPS asking for specifying your home directory and your cloned git repository root.

danielratiu, 2017-08-14:
this is an issue in a binary-distributed mbeddr RCP (which does not assume
any cloned mbeddr repo)

eugenschindler, 2017-08-14:
Ah, ok, thx for the explanation.