mbeddr/mbeddr.core #1767: Richtext Feature Request: Allow to merge IWords

digital-ember, 2017-09-21:
Currently, someone who provides their one IWords to a Text has no clear way how they could contribute to the normalizeStructure functionality of the Text concept.
In my use case, I would like to be able to "normalize"/merge my IWords if they are siblings before an empty Word is inserted by "Text.normalizeStructure".

slisson, 2017-09-21:
You could create a subconcept of Text and override the behavior method.

digital-ember, 2017-09-22:
Not sure about that. How to override a non-virtual behavior method?
Additionally, I tried to build a fix for that (https://github.com/digital-ember/mbeddr.core/tree/feature/allowIWordMerge), only to learn that the whole normalizeStructure-method in the Text concept is only called for EditorCell_Word-cells. I yet have to understand the details, but atm I think this method is only called for certain actions on the cells used to project Word_s, right?

digital-ember, 2017-09-22:
On a side note: when I use the reflective editor, my test implementation to merge ItalicText (which is an IWord), works. When I delete a Word that is between two ItalicText nodes, they get merged as expected by my implementation:
[image]

This would be my desired behavior, but with the "regular" projection. However, I fail to get rid of the empty Word between the two IWords, and "normalizeStructure" doesn't seem to be called if I use the regular projection and delete the Word character by character.