mbeddr/mbeddr.core #1769: com.mbeddr.mpsutil.filepicker should show warning instead of error when path does not exist

arimer, 2017-09-25:
Currently there is a checkingrule in the Filepicker that is showing an error when a path does not exists. 
It should be possible to set the flag: **mayBeEmpty** to avoid this error and to show a warning instead.

Looks like the flag is already there, the only thing that is missing is the possibility to set the flag to true.

Maybe an intention would to the job @bkolb ?

coolya, 2017-09-26:
Isn't it the container that uses the file picker that should decide if the path can be empty or not? How should the user with the intention decide if it's ok or not? 

If the file picker is used to select an output location for some generated artefacts it might be ok if the path doesn't exists and the generator will simply create it. But if you select a location to an SDK that you need during compilation it might not. I think the file picker can't decide this. The container has to and then set the flag on the contained file picker instance correctly. Most likely in it's own constructor.

arimer, 2017-09-27:
That might work aswell.

closed 2025-04-07 by alexanderpann