mbeddr/mbeddr.core #1782: Tables: Padding of cell content

digital-ember, 2017-10-16:
I experience problems trying to add padding to content of cells. Assume I have a node of INamedConcept as a cell content, and I want to add `padding-top: 5 spaces` to its Editor_Cell projection.

The result looks like this:
[image]
The padding is applied to the bottom.

The other padding options seem to be fine, applying a padding styling like this, for example:
```
<no base style> {  
  padding-top : 5 spaces   
  padding-left : 3 spaces  
  padding-right : 7 spaces 
  padding-bottom : 8 spaces
}
```
leads to this projection (annotated):
[image]

Changing the `vertical-alignment` property of either the cell or the content did not seem to have any effect.

slisson, 2017-11-08:
As an alternative import the language de.itemis.mps.editor.celllayout and use the margin styles.

alexanderpann, 2023-02-19:
I am closing this issue because this behavior is not reproducible for me anymore. Please open an issue in [MPS-Extensions](https://github.com/JetBrains/MPS-extensions/issues) in case you are still affected by this and have a minimal reproducible example. Two working examples with:
```
padding-top : 5 spaces   
padding-left : 3 spaces  
padding-right : 7 spaces 
padding-bottom : 1 spaces
```

[image]

[image]

closed 2023-02-19 by alexanderpann