mbeddr/mbeddr.core #1786: Errors in DocumentConfig

josefmueller, 2017-10-25:
Hi all,

If create RequirementsModule, it automatically creates DocumentConfig but there is following error shown:

[image]

There is no way to add defaultImagePath.

If I create a new DocumentConfig manually following error is shown:

[image]

I am not sure about which folder is this error talking about.

coolya, 2017-11-08:
This user shouldn't be able to pic this location anyway because the generator will override it. I think we should remove this child and find a different solution.

alexanderpann, 2023-12-15:
The original issue is now fixed in maintenance/mps20222 in https://github.com/mbeddr/mbeddr.core/commit/4313c96d94483446a63046e89408daab7eaade8e

commit 615a466e (2017-11-08): Fixed #1786 Errors in DocumentConfig

There was a child 'defaultImagePath' in DocumentConfig that was missing
in the editor. Errors for this child were shown on the parent.
closed 2023-12-15 by alexanderpann