mbeddr/mbeddr.core #1788: Parameters and variable names starting with fo cannot be typed in

eugenschindler, 2017-10-25:
How to reproduce:
Try to type in a normal mbeddr module things like this:
 double fork; // the fo will be a variable and the rk part will be unbound
 void f(double format) // again the fo will be a parameter and the rmat part will be unbound

Expected would be that one can use names for variables and parameters that start with fo (like in the examples above).

danielratiu, 2017-11-07:
>>void f(double format) // again the fo will be a parameter and the rmat part will be unbound
this works here

the case with global variable is a deeper issue ... assigned it to the editors guru

aswinsugumaran, 2017-11-07:
Similar issue inside Struct unable to create a member name *constantVar*

commit 3937f341 (2017-11-07): Fixed #1788 Parameters and variable names starting with fo cannot be typed in
closed 2017-11-07 by slisson