mbeddr/mbeddr.core #1835: DVD drive disk is available in completion menu of FileSystemDirPicker

ksun-x, 2017-12-15:
Currently, all locally available disks/drives are shown in the completion menu of FileSystemDirPicker. Nevertheless, the drive-based path is invalid, since **file.getCanonicalPath()** will cause an IO exception: 
> java.io.IOException: The device is not ready

The following screenshot shows that DVD drive **D:** is in the completion menu
[image]

ksun-x, 2017-12-18:
In addition, please make the **AbstractPicker.proposals()** method **virtual** to allow overriding

alexanderpann, 2025-03-12:
> In addition, please make the **AbstractPicker.proposals()** method **virtual** to allow overriding

This part is at least fixed.