mbeddr/mbeddr.core #1838: Parallel build in mbeddr not working

eugenschindler, 2017-12-18:
Just migrated to MPS 2017.3-based mbeddr. Default number of cores in build configuration is set to 7. This makes the build hang without ever returning.
If I set the number of cores to 1 (leading to a make -j1 instead of make -j7), then there is no problem. Tried to check if this behaves different on the command line, but there a make -j7 also gets stuck. Must be something in the makefile, but I don't know what (yet).

krishnanm86, 2017-12-18:
Hi Eugen,
        Could you please attach a zip of the generated sources to reproduce the bug and find out the source of the issue.

krishnanm86, 2018-01-24:
Closing this after fixing it offline. The issue is related to gnu make 3.81 and is a known issue with parallel makes using this make.

closed 2018-01-24 by krishnanm86

coolya, 2018-01-24:
could we do a version detection in the make facet and show an error to the user if we detect that faulty make version?

krishnanm86, 2018-02-02:
This should be possible. Created a new issue #1865