mbeddr/mbeddr.core #1853: Issue finding text through SearchPanel in a Word

ftomassetti, 2018-01-22:
The issue I am experiencing is that when searching for a text in a Word (de.slissons.mps.richtext language) the user try to serarch for "firstword secondword" with a space between two words and this does not work. However searching for "firstwordsecondword" works. I am referring to this panel:

[image]

Unfortunately I am using an old version of mbeddr (maintenance-mps34.1.0.17905.d048a69) but I did not find any reference to this being changed in recent versions so probably this problem is still present.

slisson, 2018-01-23:
#1847

ftomassetti, 2018-01-23:
Sorry I missed the related issue. I was using terms like "SearchPanel" and "find text", I did not think about RichText or copy pasting

slisson, 2018-01-23:
This was a note to myself. I'm not sure yet if they are related.

ftomassetti, 2018-01-23:
I am trying to understand a little more about this issue.

I think the text that the SearchPanel see is obtained in SearchPanel.allCellsAndContent:

```
  private Pair<List<EditorCell_Label>, String> allCellsAndContent() {
    StringBuilder sb = new StringBuilder();
    List<EditorCell_Label> cells = new ArrayList<>();
    EditorCell rootCell = myEditor.getRootCell();
    if (rootCell instanceof EditorCell_Label) {
      EditorCell_Label cell_label = (EditorCell_Label) rootCell;
      cells.add(cell_label);
      sb.append(cell_label.getRenderedText());
    }
    if (rootCell instanceof EditorCell_Collection) {
      EditorCell_Collection collection = (EditorCell_Collection) rootCell;
      List<EditorCell_Label> editorCell_labelList = CollectionUtil.filter(EditorCell_Label.class, collection.dfsCells());
      for (EditorCell_Label label : editorCell_labelList) {
        if (PunctuationUtil.hasLeftGap(label)) {
          sb.append(' ');
        }
        sb.append(label.getRenderedText());
      }
      cells.addAll(editorCell_labelList);
    }
    return new Pair<>(cells, sb.toString());
  }
```

The issue seems to be in the Text editor. 

The Text cell is a EditorCell_Collection and running this code I get a string in which the different words are not separated by spaces:

```
EditorCell cellForText = editorContext.getEditorComponent().findNodeCell(node.text); 
print ( "CellForText: "+ cellForText ) ; 
EditorCell_Collection collection = (EditorCell_Collection) cellForText; 
StringBuilder sb = new StringBuilder(); 
list<EditorCell_Label> editorCell_labelList = (list<EditorCell_Label>) CollectionUtil.filter(EditorCell_Label.class, collection.dfsCells()); 
for (EditorCell_Label label : editorCell_labelList) { 
  if (PunctuationUtil.hasLeftGap(label)) { 
    sb.append(' '); 
  } 
  sb.append(label.getRenderedText()); 
} 
print ( "Obtained: "+ sb.toString() ) ;
```

How this could be solved?
* Adding a space into the text returned by getRenderedText for the cell of the single words?
* Adding cells that represents spaces and are not selectable?
* Finding a way to make PunctuationUtil.hasLeftGap return true for the word cells?

ftomassetti, 2018-01-23:
Looking in MPS source code it seems that `dfsCells` is used only in this context ([https://github.com/JetBrains/MPS/search?utf8=%E2%9C%93&q=dfsCells&type=](https://github.com/JetBrains/MPS/search?utf8=%E2%9C%93&q=dfsCells&type=)), so maybe what we could do is to override it to return some extra cells that are not actually displayed in the editor.

ftomassetti, 2018-05-25:
I just checked and the issue is still present with he last version of mbeddr platform.
I realize there are over 400 isues open at the moment but maybe with some directions I could help on this one

closed 2018-05-29 by slisson