mbeddr/mbeddr.core #1854: Unit evaluation in math power function

jackkandelaars, 2018-01-24:
I have a problem with the evaluation of the units when using the math power function.
The following units I have defined:
[image]

I use these units in the following  calculation. GetVOut returns a double/mmps/ type. See next figure.
[image]

The math pow function returns a double type. See next figure.
[image]

The exponent of the power function is an integer so I didn’t expect a double but a double/mm2ps2/ type. See next figure.
[image]

I understand that with an floating point exponent you can’t determine the unit of the result but for a integer exponent you would expect an unit result.
Is the above statement correct? If yes can this issue be corrected?

krishnanm86, 2018-02-01:
#1864

alexanderpann, 2022-04-27:
Closing because it seems to be fixed in https://github.com/mbeddr/mbeddr.core/pull/1864.

closed 2022-04-27 by alexanderpann