mbeddr/mbeddr.core #1855: Dataflow not correct with math functions

jackkandelaars, 2018-01-24:
I have the following code snippet where a warning is issued that the variable is not used. The variable is used in the frac math expression so I don't expect this warning. The sqrt math expression has the same problem.

[image]

If I use the variable in the regular mbeddr code I don't get this warning.

[image]

krishnanm86, 2018-02-13:
@jackkandelaars The fix is being reviewed in the following pull request #1871

krishnanm86, 2018-02-15:
Issue fixed and closed

closed 2018-02-15 by krishnanm86