mbeddr/mbeddr.core #1857: Preferences Panel appearing only if actions filter is present

ftomassetti, 2018-01-25:
Something very weird is happening. If I define a preference panel it does not appear among preferences UNLESS I add in the same plugin an actions filter (even empty). I tried that again and again: if I remove the actions filter the preferences panel disappear. I guess that the preferences panel has the need for some library at runtime that is not there unless I add the actions filter

**Beware, I am using an old version of the mbeddr platfom: maintenance-mps34.1.0.17905.d048a69**

alexanderpann, 2025-03-12:
Let's keep the issue open, although I'll really like to close it because the MPS version is so old. Very unlikely that this issue still exists.

alexanderpann, 2025-04-07:
On second thought, this issue was not reported once in the last years and the MPS version is too old. A lot of refactorings were also done in regards to class loading in recent MPS versions.

closed 2025-04-07 by alexanderpann