mbeddr/mbeddr.core #1858: Copy/paste node as XML doesn't work on multiple nodes

eugenschindler, 2018-01-25:
When selecting a number of nodes, I would expect that I can copy them as XML, but that doesn't work. It only works when clicking one single node.

coolya, 2018-01-26:
What do you expect to be copied in that case? The two nodes next to each other without a wrapper or with a wrapper:

```
<node A>
...
</node A>
<node B>
...
</node B>
```

vs.
```
<nodes>
    <node A>
    ...
    </node A>
    <node B>
    ...
    </node B>
</nodes>
```

The first will cause some problems because most XML parses will refuse to parse it cause they expect a document to start with a single root. This would also cause problem with the "Paste XML" action which allow you to paste nodes from xml in the clipboard.

eugenschindler, 2018-01-27:
Hi Kolja, thanks for reacting so quickly! The main use case that I would use this for is the "Paste XML" case. I can give a simple example: let's say I have a build solution and I want to copy-paste all the macros that are defined on top of the solution to another build solution. To complicate things a bit, I don't do this within the same MPS instance, but let's say the one I want to paste to runs on a remote machine (but I can reach its GUI via remote desktop or SSH terminal). If I want to select all the folder macros, I will have an nlist. The node above that is immediately the entire build solution.
Now, I know that you can have a number of workarounds for this example, but the point is: I think usability for copy-pasting parts of a model between MPS isntances will be helped a lot if you can copy-paste nlists.
Technically, I could imagine that you can always put a wrapper around copyied nlist (like in the second case you describe), but the "Paste XML" functionality will need to know somehow to deal with it.