mbeddr/mbeddr.core #1870: updateDescription runs in RCP, but not in DevEnv

RobertWalter83, 2018-02-13:
Hi,
we make use of the IntentionGroupAnnotation to group our intentions.
This works fine in our development environment.

However, in the RCP we build, the group name is not only used as a is used as a separator in the menu, but also as a prefix for each intention. 
The only place we found could be responsible for this is the script "updateDescription" in the mpsutil.intentions language.
We try to understand why this script even exists and why it is seemingly not called in our DevEnv?

We experience a similar issue in the ContextActions ToolWindow, where intentions are also shown as "group.label + intention.description()", whereas we would prefer to not have the group name displayed.
For non-parameterized intentions, we can override the label by setting the label of the IntentionAnnotation itself, but we cannot seem to access the parameter in case of parameterized intentions in the label-field ("not applicaple in this context") which we require to have a parameterized label.

slisson, 2018-02-13:
This is what the IntentionGroupAnnotation actually does. It adds the group name as a prefix to the intentions name separated with ": ".

The original MPS intentions menu is replaced by the one in com.mbeddr.mpsutil.intentions.runtime, which creates the groups out of this naming convention.

RobertWalter83, 2018-02-13:
Okay, so, how come the intention menu for a node invoked in our RCP looks like this (which meets the expectation, if I understand you correctly):
[image]
While the same menu on the same node in our DevEnv looks like this:
[image]

Still confused...?

RobertWalter83, 2018-02-13:
Also, is there a way that would allow us to avoid having the group label as a prefix of the intention description? We would like to just use the separator, if possible. If not, I will create a feature request. Thanks!

slisson, 2018-02-13:
The prefix is removed from the label after putting it into a group. The second picture is the expected result. I have no idea how the first picture is possible.

RobertWalter83, 2018-02-13:
Btw: in OriginalIntentionMenu are some System.err.println statements that you might want to remove before the next release.

alexanderpann, 2024-07-14:
The issue in the first screenshot was never reported again, so I'll close the ticket. The System.err.println statements were also removed a long time ago.

closed 2024-07-14 by alexanderpann