mbeddr/mbeddr.core #1872: IVisibleElementAdapter could return a list of nodes

AlexeiQ, 2018-02-13:


joao-silveira, 2018-02-13:
We found a case where a specific chunk content, should expose its own content as public to other chunks.

Currently as a solution one could make use of the Chunk concept interface method: 
`public virtual abstract sequence<node<>> allReferenceableContentsInChunk();`
To expose not only this given specific chunk, but also its own content.
                                                                          
From my point of view it would be a much cleaner solution if we could simply reuse IVisibleElementAdapter changing its interface to return a list of nodes instead of single node.

coolya, 2018-02-13:
I think the chunk way is the right way to do it anyway. This is how we implemented it in the past in other projects, including mbeddr.

joao-silveira, 2018-02-13:
Can you point me to an example in mbeddr? couldn't find any similar one.

coolya, 2018-02-13:
`Section` and `Module` for instance.