mbeddr/mbeddr.core #1891: Refactor MPS node comparator

szabta89, 2018-03-19:
The current implementation uses a lot of deprecated MPS code. Let's get rid of them. 
While doing the refactoring, we should also take care of the following things:
* Instead of allowing to pass in sets of `IgnoredProperty`s and `IgnoredReference`s, we should simply rely on one set that contains `SConceptFeature`s. 
* For compatibility with legacy code, we could keep around the two classes `IgnoredProperty` and `IgnoredReference`, but the "real" implementation should simply accept a `set<SConceptFeature>`. Of course, this means that the `SConceptFeature`s must be obtained from the legacy versions of the method.

coolya, 2018-03-29:
I addition we should overhaul the error messages so that clearly stat what was expected and the actual node and have them in a more structured way. They currently read like sentence and not like structured error message.

alexanderpann, 2022-04-27:
Closing because seems to be fixed in https://github.com/mbeddr/mbeddr.core/commit/948d044d06dedeb5fbbf0cf98e758b14152c1ada.

closed 2022-04-27 by alexanderpann