mbeddr/mbeddr.core #1930: Editor exception depending on resolution

ftomassetti, 2018-06-08:
We have a problem appearing on Windows when using a certain resolution (1920x1200) but not other resolutions (like 1920x1080). We are using mbeddr 1.0.20315.2a06849

This is the exception we got in an editor that fails to render:
```
RUNNING SEQUENCE DiarrheaScenarios index=116, total=116, readOnly=false
  isRunning: true
(sequence completed)
[4370813]   INFO - .diagnostic.PerformanceWatcher - High memory usage (free 142 of 2014 MB) while dumping threads to C:\Users\laurent.wiart\.MPS2018.1\system\log\threadDumps-freeze-20180608-122721-MPS-181.1367\threadDump-20180608-122727.txt
[4372563]   INFO - .openapi.util.LowMemoryWatcher - Low memory signal received.
[4374939]  ERROR -  jetbrains.mps.ide.ThreadUtils - jetbrains.mps.ide.ThreadUtils
java.lang.RuntimeException: Cannot set width to 924. Maximum width is 10
        at de.itemis.mps.editor.celllayout.runtime.LayoutableAdapter.setSize(LayoutableAdapter.java:56)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableAdapter.setHeight(LayoutableAdapter.java:99)
        at de.itemis.mps.editor.celllayout.layout.Grid.growElement(Grid.java:370)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:61)
        at de.itemis.mps.editor.celllayout.layout.GridLayout.layout(GridLayout.java:22)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.layout.GridLayout.layout(GridLayout.java:22)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.layout.AbstractGridLayout.layout(AbstractGridLayout.java:65)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.relayout(LayoutableCollectionAdapter.java:146)
        at de.itemis.mps.editor.celllayout.runtime.LayoutEngine$2.run(LayoutEngine.java:90)
        at de.itemis.mps.editor.celllayout.runtime.LayoutWatchdog.enter(LayoutWatchdog.java:22)
        at de.itemis.mps.editor.celllayout.runtime.LayoutEngine.layoutSubtreeUnsafe(LayoutEngine.java:80)
        at de.itemis.mps.editor.celllayout.runtime.LayoutEngine.layoutSubtree(LayoutEngine.java:75)
        at de.itemis.mps.editor.celllayout.runtime.LayoutEngine.layoutSubtree(LayoutEngine.java:57)
        at de.itemis.mps.editor.celllayout.runtime.LayoutEngine.layoutCell(LayoutEngine.java:53)
        at de.itemis.mps.editor.celllayout.runtime.LayoutInterceptor.doLayout(LayoutInterceptor.java:62)
        at jetbrains.mps.nodeEditor.cells.EditorCell_Collection.relayoutImpl(EditorCell_Collection.java:515)
        at jetbrains.mps.nodeEditor.cells.EditorCell_Basic.relayout(EditorCell_Basic.java:880)
        at jetbrains.mps.nodeEditor.EditorComponent.doRelayout(EditorComponent.java:1656)
        at jetbrains.mps.nodeEditor.EditorComponent.relayout(EditorComponent.java:1643)
        at jetbrains.mps.nodeEditor.EditorContext.lambda$restoreEditorComponentState$0(EditorContext.java:219)
        at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:71)
        at jetbrains.mps.nodeEditor.EditorContext.restoreEditorComponentState(EditorContext.java:218)
        at jetbrains.mps.nodeEditor.updater.UpdaterImpl.doUpdate(UpdaterImpl.java:116)
        at jetbrains.mps.nodeEditor.updater.UpdaterImpl.update(UpdaterImpl.java:91)
        at jetbrains.mps.nodeEditor.EditorComponent.rebuildEditorContent(EditorComponent.java:1792)
        at jetbrains.mps.nodeEditor.NodeEditorComponent.rebuildEditorContent(NodeEditorComponent.java:139)
        at com.voluntis.martian.expr.testing.behavior.AlgoTestCoverageAssQuery__BehaviorDescriptor.runQuery_id7hIyKqbFNeu(AlgoTestCoverageAssQuery__BehaviorDescriptor.java:112)
        at com.voluntis.martian.expr.testing.behavior.AlgoTestCoverageAssQuery__BehaviorDescriptor.invokeSpecial0(AlgoTestCoverageAssQuery__BehaviorDescriptor.java:201)
        at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invokeSpecial(BaseBHDescriptor.java:346)
        at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invokeVirtual(BaseBHDescriptor.java:305)
        at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invoke(BaseBHDescriptor.java:229)
        at jetbrains.mps.core.aspects.behaviour.SMethodImpl.invoke0(SMethodImpl.java:151)
        at jetbrains.mps.core.aspects.behaviour.SMethodImpl.invoke(SMethodImpl.java:126)
        at com.mbeddr.core.base.behavior.AssessmentQuery__BehaviorDescriptor.executeQuery_idK292flwJAp(AssessmentQuery__BehaviorDescriptor.java:60)
[... 54 lines cut ...]
        at com.intellij.openapi.command.impl.CoreCommandProcessor.executeCommand(CoreCommandProcessor.java:85)
        at jetbrains.mps.smodel.WorkbenchModelAccess.runWriteActionInCommand(WorkbenchModelAccess.java:311)
        at jetbrains.mps.smodel.WorkbenchModelAccess.executeCommand(WorkbenchModelAccess.java:307)
        at jetbrains.mps.smodel.ModelAccess.executeCommand(ModelAccess.java:122)
        at com.voluntis.martian.statemachine.simtest.behavior.MartianScenario__BehaviorDescriptor$13.run(MartianScenario__BehaviorDescriptor.java:308)
        at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:762)
        at java.awt.EventQueue.access$500(EventQueue.java:98)
        at java.awt.EventQueue$3.run(EventQueue.java:715)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:732)
        at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:779)
[... cut ...]
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
  concreteRunning:afterSimulating FM:306525312

```

ftomassetti, 2018-06-08:
Update: probably not connected to resolution but to another settings (the show interpreter coverage being active/not active)

alexanderpann, 2024-07-14:
This error was reduced to a warning a long time ago. It shouldn't happen though, so please open a new issue when you still see the message.

closed 2024-07-14 by alexanderpann