mbeddr/mbeddr.core #1935: IfExpression having both elsePart and elseExpression cause coverage issue

ftomassetti, 2018-06-22:
In a project an IfExpression had both an elsePart and an elseExpression populated. The problem is that it was visible only using the reflective editor and no error was displayed. This fact was noticed because the coverage was not 100% because of this.

krishnanm86, 2018-06-25:
Hi @ftomassetti ,
     Could you provide some sample code/minimum example.

ftomassetti, 2018-06-25:
I do not think would be needed: very simply an IfExpression can have both a child under the role `elsePart` and a child under the role `elseExpression`. If both the roles are not null what happens is that just one of them is displayed in the default editor and only one of them is considered during coverage, leading to a coverage below 100%. I would suggest that it should be an error to populate the two parts at the same time and that it should be possible to view both parts in the default editor if both parts are present.

ftomassetti, 2018-06-25:
To reproduce it you can use the reflective editor and populate both elsePart and elseExpression