mbeddr/mbeddr.core #1940: Issue with MartianPlantUMLSVGDocumentLoader

ftomassetti, 2018-06-26:
After updating to a more recent version of MPS and the mbeddr platform we have started facing an issue with the SVG panel.

```
java.lang.NoClassDefFoundError: Could not initialize class org.apache.batik.bridge.BridgeContext
	at org.apache.batik.swing.svg.AbstractJSVGComponent.createBridgeContext(AbstractJSVGComponent.java:869)
	at org.apache.batik.swing.svg.AbstractJSVGComponent.installSVGDocument(AbstractJSVGComponent.java:739)
	at org.apache.batik.swing.JSVGCanvas.installSVGDocument(JSVGCanvas.java:591)
	at org.apache.batik.swing.svg.AbstractJSVGComponent$2.run(AbstractJSVGComponent.java:679)
	at org.apache.batik.swing.svg.AbstractJSVGComponent.stopThenRun(AbstractJSVGComponent.java:708)
	at org.apache.batik.swing.svg.AbstractJSVGComponent.setSVGDocument(AbstractJSVGComponent.java:677)
	at org.apache.batik.swing.svg.AbstractJSVGComponent$SVGListener.documentLoadingCompleted(AbstractJSVGComponent.java:1451)
	at com.mbeddr.mpsutil.plantuml.pluginSolution.plugin.PlantUMLSVGDocumentLoader$2.dispatch(PlantUMLSVGDocumentLoader.java:26)
	at org.apache.batik.util.EventDispatcher.dispatchEvent(EventDispatcher.java:103)
	at org.apache.batik.util.EventDispatcher.fireEvent(EventDispatcher.java:87)
	at org.apache.batik.util.EventDispatcher$1.run(EventDispatcher.java:46)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:301)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:762)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:732)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:779)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:720)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:395)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

stephaneberle9, 2018-06-27:
Yes, we are also seeing this exception. And sometimes we also see the following `Caused by` trace right behind:

```
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassCastException: org.apache.batik.bridge.RhinoInterpreterFactory cannot be cast to org.apache.batik.script.InterpreterFactory
        at org.apache.batik.script.InterpreterPool.<clinit>(InterpreterPool.java:66)
        ... 29 more
```

This is not a "normal" class cast exception though, because the `RhinoInterpreterFactory` did implement the `org.apache.batik.script.InterpreterFactory` interface before and does continue to do so now. However, such kind of class cast exceptions can happen anyway in cases where the JAR file containing the `org.apache.batik.script.InterpreterFactory` interface (i.e., `batik-script.jar`) is loaded twice. Obviously, the `batik-script.jar` file is present once in the `com.mbeddr.mpsutil.plantuml.pluginSolution`.

Given these facts, I was wondering if the same JAR file also exists somewhere else in the new MPS version or in a plugin that comes along with it. Maybe, its also an accidental duplication caused by the move the the mbeddr platform to MPS extensions. I've tried to go to the bottom of this but wasn't able to find anything suspicious or any sort of confirmation for these assumptions so far. But maybe/hopefully these thoughts will let ring the alarm bells in anyone else's head ;)

slisson, 2018-07-23:
The RhinoInterpreterFactory was moved to a different package in newer versions. Now the class from the MPS lib folder is not shadowed anymore by the class in the solution. Updating the batik library should fix this.

slisson, 2018-07-25:
It fails with MPS 2018.2 RC1. Reverted the change on master.

alexanderpann, 2024-03-18:
This is not an issue anymore.

commit 1ffe51ab (2018-07-24): Update batik library in mpsutil.plantuml

Fixes #1940
closed 2024-03-18 by alexanderpann