mbeddr/mbeddr.core #1995: update failed for ActionGroup: Print Editor ()[Print Editor]

KemalSoysal, 2018-10-04:
When using the MPS 2018.2.3 from sources with the plugins of mbeddr I get this exception:

update failed for ActionGroup: Print Editor ()[Print Editor]

jetbrains.mps.classloading.ModuleClassLoader$ModuleClassLoaderIsDisposedException: ClassLoader of the module 'com.mbeddr.core.base.pluginSolution [solution]' is disposed and not operable!
	at jetbrains.mps.classloading.ModuleClassLoader.checkNotDisposed(ModuleClassLoader.java:87)
	at jetbrains.mps.classloading.ModuleClassLoader.loadClass(ModuleClassLoader.java:115)
	at jetbrains.mps.classloading.ModuleClassLoader.loadClass(ModuleClassLoader.java:111)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at com.mbeddr.core.base.pluginSolution.plugin.screenshotNode_Action.collectActionData(screenshotNode_Action.java:50)
	at jetbrains.mps.workbench.action.BaseAction$CollectActionData.compute(BaseAction.java:245)
	at jetbrains.mps.workbench.action.BaseAction$1.run(BaseAction.java:124)
	at jetbrains.mps.smodel.WorkbenchModelAccess.lambda$runReadAction$0(WorkbenchModelAccess.java:77)
	at com.intellij.openapi.application.impl.ApplicationImpl.runReadAction(ApplicationImpl.java:925)
	at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:74)
	at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:53)
	at jetbrains.mps.workbench.action.BaseAction.update(BaseAction.java:121)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performDumbAwareUpdate(ActionUtil.java:136)
	at com.intellij.openapi.actionSystem.impl.Utils.doUpdate(Utils.java:256)
	at com.intellij.openapi.actionSystem.impl.Utils.expandActionGroup(Utils.java:199)
	at com.intellij.openapi.actionSystem.impl.Utils.expandActionGroup(Utils.java:225)
	at com.intellij.openapi.actionSystem.impl.ActionToolbarImpl.updateActionsImpl(ActionToolbarImpl.java:1100)
	at com.intellij.openapi.actionSystem.impl.ActionToolbarImpl.access$000(ActionToolbarImpl.java:53)
	at com.intellij.openapi.actionSystem.impl.ActionToolbarImpl$2.updateActionsImpl(ActionToolbarImpl.java:175)
	at com.intellij.openapi.actionSystem.impl.ToolbarUpdater$MyUpdateRunnable.run(ToolbarUpdater.java:186)
	at com.intellij.util.ui.UIUtil.invokeLaterIfNeeded(UIUtil.java:3097)
	at com.intellij.ide.IdeEventQueue.ifFocusEventsInTheQueue(IdeEventQueue.java:183)
	at com.intellij.ide.IdeEventQueue.executeWhenAllFocusEventsLeftTheQueue(IdeEventQueue.java:132)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.doWhenFocusSettlesDown(FocusManagerImpl.java:190)
	at com.intellij.openapi.actionSystem.impl.ToolbarUpdater.updateActions(ToolbarUpdater.java:108)
	at com.intellij.openapi.actionSystem.impl.ToolbarUpdater.access$400(ToolbarUpdater.java:40)
	at com.intellij.openapi.actionSystem.impl.ToolbarUpdater$MyTimerListener.run(ToolbarUpdater.java:158)
	at com.intellij.openapi.actionSystem.impl.WeakTimerListener.run(WeakTimerListener.java:54)
	at com.intellij.openapi.actionSystem.impl.ActionManagerImpl$MyTimer.runListenerAction(ActionManagerImpl.java:1428)
	at com.intellij.openapi.actionSystem.impl.ActionManagerImpl$MyTimer.notifyListeners(ActionManagerImpl.java:1417)
	at com.intellij.openapi.actionSystem.impl.ActionManagerImpl$MyTimer.actionPerformed(ActionManagerImpl.java:1401)
	at javax.swing.Timer.fireActionPerformed(Timer.java:313)
	at javax.swing.Timer$DoPostEvent.run(Timer.java:245)
	at java.awt.event.InvocationEvent.dispatch$$$capture(InvocationEvent.java:311)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:781)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:722)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:382)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

KemalSoysal, 2018-10-04:
happens also in iets
[9371720]  ERROR - penapi.actionSystem.impl.Utils - update failed for ActionGroup: Clear Inspectors ()[Clear Inspectors] 
jetbrains.mps.classloading.ModuleClassLoader$ModuleClassLoaderIsDisposedException: ClassLoader of the module 'org.iets3.core.expr.plugin [solution]' is disposed and not operable!
	at jetbrains.mps.classloading.ModuleClassLoader.checkNotDisposed(ModuleClassLoader.java:87)
	at jetbrains.mps.classloading.ModuleClassLoader.loadClass(ModuleClassLoader.java:115)
	at jetbrains.mps.classloading.ModuleClassLoader.loadClass(ModuleClassLoader.java:111)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at org.iets3.core.expr.plugin.plugin.deleteValuesInspectors_Action.collectActionData(deleteValuesInspectors_Action.java:39)
	at jetbrains.mps.workbench.action.BaseAction$CollectActionData.compute(BaseAction.java:245)
	at jetbrains.mps.workbench.action.BaseAction$1.run(BaseAction.java:124)
	at jetbrains.mps.smodel.WorkbenchModelAccess.lambda$runReadAction$0(WorkbenchModelAccess.java:77)
	at com.intellij.openapi.application.impl.ApplicationImpl.runReadAction(ApplicationImpl.java:925)
	at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:74)
	at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:53)
	at jetbrains.mps.workbench.action.BaseAction.update(BaseAction.java:121)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performDumbAwareUpdate(ActionUtil.java:136)
	at com.intellij.openapi.actionSystem.impl.Utils.doUpdate(Utils.java:256)
	at com.intellij.openapi.actionSystem.impl.Utils.expandActionGroup(Utils.java:199)
	at com.intellij.openapi.actionSystem.impl.Utils.expandActionGroup(Utils.java:225)
	at com.intellij.openapi.actionSystem.impl.ActionToolbarImpl.updateActionsImpl(ActionToolbarImpl.java:1100)
	at com.intellij.openapi.actionSystem.impl.ActionToolbarImpl.access$000(ActionToolbarImpl.java:53)
	at com.intellij.openapi.actionSystem.impl.ActionToolbarImpl$2.updateActionsImpl(ActionToolbarImpl.java:175)
	at com.intellij.openapi.actionSystem.impl.ToolbarUpdater$MyUpdateRunnable.run(ToolbarUpdater.java:186)
	at com.intellij.util.ui.UIUtil.invokeLaterIfNeeded(UIUtil.java:3097)
	at com.intellij.ide.IdeEventQueue.ifFocusEventsInTheQueue(IdeEventQueue.java:183)
	at com.intellij.ide.IdeEventQueue.executeWhenAllFocusEventsLeftTheQueue(IdeEventQueue.java:132)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.doWhenFocusSettlesDown(FocusManagerImpl.java:190)
	at com.intellij.openapi.actionSystem.impl.ToolbarUpdater.updateActions(ToolbarUpdater.java:108)
	at com.intellij.openapi.actionSystem.impl.ToolbarUpdater.access$400(ToolbarUpdater.java:40)
	at com.intellij.openapi.actionSystem.impl.ToolbarUpdater$MyTimerListener.run(ToolbarUpdater.java:158)
	at com.intellij.openapi.actionSystem.impl.WeakTimerListener.run(WeakTimerListener.java:54)
	at com.intellij.openapi.actionSystem.impl.ActionManagerImpl$MyTimer.runListenerAction(ActionManagerImpl.java:1428)
	at com.intellij.openapi.actionSystem.impl.ActionManagerImpl$MyTimer.notifyListeners(ActionManagerImpl.java:1417)
	at com.intellij.openapi.actionSystem.impl.ActionManagerImpl$MyTimer.actionPerformed(ActionManagerImpl.java:1401)
	at javax.swing.Timer.fireActionPerformed(Timer.java:313)
	at javax.swing.Timer$DoPostEvent.run(Timer.java:245)
	at java.awt.event.InvocationEvent.dispatch$$$capture(InvocationEvent.java:311)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:781)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:722)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:382)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

alexanderpann, 2025-04-10:
I'll close this as not reproducible anymore. Please reopen the ticket if it still happens.

closed 2025-04-10 by alexanderpann