mbeddr/mbeddr.core #2015: Prevent name mangling option and external APIs

BjoernLange, 2018-11-20:
The [mbeddr User Guide](http://mbeddr.com/userguide/UserGuideExport.html) says in [section 5.2](http://mbeddr.com/userguide/UserGuideExport.html#sid3082376456307360126):

> The second option is to select the prevent name mangling configuration item in the build configuration. This will try to get rid of the prefixes where possible, i.e., if an only if a non-prefixed name is globally unique within an executable. This optimization is performed each time you rebuild your code.

However, it is possible to define a function with the same name as an external function in mbeddr code and its name will not be mangled although it would be required to generate valid code. An example for this behavior is `com.mbeddr.core.unittest.runtime.UnitTestUtil` which defines the `strrchr` function that collides with the `strrchr` function from `string.h`. In my opinion, the `strrchr` function name should be mangled in this case (according to the quote given above).

stephaneberle9, 2018-11-20:
Hi Björn,

Yes, this looks like a bug and should be recorded as such (i.e., GitHub
issue).

Btw, the specific case of the conflicting strrchr function in
com.mbeddr.core.unittest.runtime.UnitTestUtil has already been addressed by
renaming the very same function to strreplchr.

Regards,

Stephan

stephaneberle9, 2018-11-20:
Yes, this looks like a valid error that must be corrected bug.

Btw, the specific case of the conflicting `strrchr` function in `com.mbeddr.core.unittest.runtime.UnitTestUtil `has already been addressed by renaming the very same function to `strreplchr`.