mbeddr/mbeddr.core #2042: m-Better packaging for mbeddr platform

ty1824, 2019-01-30:
We have a number of projects depending on the mbeddr platform using various ranges of the functionality. In several of those projects, we only depend on a very small set of the available tooling, but still have to include the entire platform in our project, and all of the tools that we don't use still are loaded during build, etc.

It would be very nice if each tool were published independently. Is this something that could be done?

coolya, 2019-02-26:
I would love to make this happen but it's quite a lot of effort to do this. 

The major problem I don't have viable solution for so far is, that when publish each idea plugin of the platform as a individual maven artifact we have export the dependencies of these plugins from MPS to gradle/maven so that we can make them part  of the POM. That way clients that depend on a specific plugin could download that plugin plus all of it's (transitive) dependencies. I have not yet found a way to do this in an automated way. Ideally we would have generator that would generated the required gradle scripts automatically during the generation of the ANT scripts from the build models. 

Creating these publication manually and mirroring the dependencies by hand isn't really possible given the large amount of individual plugins that currently ship in the platform. Ideally we would have generic solution in the mps-gradle-plugin for the gradle part and some extension to the build language that allows us to export this information from MPS build scripts.

I will take this idea  into our next MPS hackathon week but I can't promise any results.

ty1824, 2019-02-26:
I agree - I can see that it'd be a monumental effort to build/maintain all of the publishing for that. We've been experimenting with ways to configure MPS with gradle - i.e. use Gradle to pull in dependencies and then include them in MPS in various ways. Currently we auto-generate Solutions and then auto-add entries for them in the `libraries.xml` file. Ideally we'll be able to load them as virtual solutions and not require any additions to source besides `build.gradle`.

Perhaps an approach like this would be helpful, as dependencies declared in a Gradle build could be automatically included in a POM generated by that build. I believe there are plugins that already do this.

> Ideally we would have generator that would generated the required gradle scripts automatically during the generation of the ANT scripts from the build models.

That's related to another thing I'd also love to see, which is a gradle version of the MPS build scripts. Those ANT files are scary.