mbeddr/mbeddr.core #2057: com.mbeddr.mpsutil.conceptdiagram: Wrong notation for rendered references

arimer, 2019-04-02:
**Aggregation used for rendering references**
The concept diagram is currently using a wrong presentation for the rendering of references. 
In the example below, a simple cross reference is rendered as an [Aggregation relation](https://en.wikipedia.org/wiki/Object_composition#Aggregation) between two concepts.
[image]

felixreimann, 2019-06-30:
Isn't this correct? According to https://en.wikipedia.org/wiki/Object_composition#UML_notation
* *composition* (filled diamond): strong "has a" relation, i.e., the component ceases to exist if the composition object is deleted. This corresponds to the *child* relation in MPS.
* *aggregation* (unfilled diamond): weak "has a" relation, i.e., the component may outlive the aggregate object. This corresponds to the *references* relation in MPS.

eugenschindler, 2021-05-11:
@arimer so is it correct? If it is, maybe good to clean up this issue?

closed 2021-05-11 by arimer