mbeddr/mbeddr.core #2066: Custom project view: preserve expansion state

szabta89, 2019-04-18:
The custom project view has some odd user experience: 
* It flickers. When you work in MPS and the custom view is active, then once in a while the view is refreshed (by something, not sure what triggers that), and this refresh is visible. 
* The expansion state is sometimes lost. 
* There is no way to select an element programmatically.

blockhart11, 2019-06-11:
To add to the list above, sometimes the navigator suddenly switches from the custom view back to the logical/default view. In my case, in MPS, it happens consistently when I create a new model through a right-click action.

slisson, 2019-06-13:
Die default view is activated at the end of `jetbrains.mps.ide.actions.NewModel_Action`. There is nothing we can do to change this behavior.

slisson, 2019-06-14:
- The projectview language now has an expression for referencing a project view. You can write this to select a node: `projectView<MyView, project>.selectNode(node, true)`.
- The expansion state and flickering issues are also fixed. Let me know if there are still cases where it doesn't work.
- The switch to the default view after a model update is something that needs to be fixed by JetBrains.

closed 2019-06-14 by slisson