mbeddr/mbeddr.core #2070: plugins dependencies to stub-models not resolved in upstream projects

arimer, 2019-05-22:
Currently we have a problem with dependency resolution in upstream projects.

[image]

Whenever a `@stub` model is packaged as part of a language which is loaded as a pluging, dependencies to the packaged stub models cannot be resolved. 
We had a similar problem when migrating [iets3.os](https://github.com/IETS3/iets3.opensource/pull/224) project to 2019.1

arimer, 2019-05-22:
Not sure if this is on purpose no longer supported by MPS, but nevertheless we should move stubs in a stub solution to avoid problems here.

arimer, 2019-05-22:
Same problem for: `httpserver.plugin`
[image]

slisson, 2019-06-07:
Solution is to mark all individual jars as source and not the lib folder. See https://github.com/JetBrains/MPS-extensions/commit/1d3e653564f9f13b4596106b2394a2aa28c676b8

commit 1b1db427 (2019-06-07): Stub models were broken in some packaged solutions

Fixes #2070
closed 2019-06-07 by slisson