mbeddr/mbeddr.core #2107: Rule in interpreter causes an NPE in unrelated piece of code

ftomassetti, 2019-11-25:
In a project I have noticed that commenting/uncommenting this rule causes an exception:

[image]

This seems true for all the rules in these languages which have constraints (i.e., they specify the type of the values they work on).

When they are active I can build the languages in MPS but the build fails from the command line

```
07:08:14]W:		 [:code:buildLanguages]  [generate] Generation failed for model 'com.voluntis.martian.versioning@descriptor', unexpected error in generator query: failed to evaluate property value
[07:08:14]W:		 [:code:buildLanguages]  [generate] -- -- was input node: [evaluators] ConceptEvaluator null[2696209964780468584] in com.voluntis.martian.expr.plugin@7_11
[07:08:14]W:		 [:code:buildLanguages]  [generate] java.lang.NullPointerException
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.lang.core.behavior.BaseConcept__BehaviorDescriptor.getPresentation_idhEwIMiw(BaseConcept__BehaviorDescriptor.java:53)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.lang.core.behavior.BaseConcept__BehaviorDescriptor.invokeSpecial0(BaseConcept__BehaviorDescriptor.java:94)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invokeSpecial(BaseBHDescriptor.java:357)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invokeVirtual(BaseBHDescriptor.java:316)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invoke(BaseBHDescriptor.java:240)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.SMethodImpl.invoke0(SMethodImpl.java:151)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.SMethodImpl.invoke(SMethodImpl.java:126)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at com.mbeddr.mpsutil.interpreter.behavior.Util.sanitizeTypeName(Util.java:62)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at com.mbeddr.mpsutil.interpreter.behavior.TypedChildConstraint__BehaviorDescriptor.getPresentation_idhEwIMiw(TypedChildConstraint__BehaviorDescriptor.java:48)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at com.mbeddr.mpsutil.interpreter.behavior.TypedChildConstraint__BehaviorDescriptor.invokeSpecial0(TypedChildConstraint__BehaviorDescriptor.java:138)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invokeSpecial(BaseBHDescriptor.java:357)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invokeVirtual(BaseBHDescriptor.java:316)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invoke(BaseBHDescriptor.java:240)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.SMethodImpl.invoke0(SMethodImpl.java:151)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.SMethodImpl.invoke(SMethodImpl.java:126)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at com.mbeddr.mpsutil.interpreter.behavior.ConceptEvaluator__BehaviorDescriptor$1.select(ConceptEvaluator__BehaviorDescriptor.java:47)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at com.mbeddr.mpsutil.interpreter.behavior.ConceptEvaluator__BehaviorDescriptor$1.select(ConceptEvaluator__BehaviorDescriptor.java:45)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.internal.collections.runtime.ISelector.invoke(ISelector.java:12)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.internal.collections.runtime.impl.SelectingSequence$SelectingIterator.moveToNext(SelectingSequence.java:60)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.internal.collections.runtime.impl.SelectingSequence$SelectingIterator.hasNext(SelectingSequence.java:32)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.internal.collections.runtime.IterableUtils.join(IterableUtils.java:88)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at com.mbeddr.mpsutil.interpreter.behavior.ConceptEvaluator__BehaviorDescriptor.getPresentation_idhEwIMiw(ConceptEvaluator__BehaviorDescriptor.java:45)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at com.mbeddr.mpsutil.interpreter.behavior.ConceptEvaluator__BehaviorDescriptor.invokeSpecial0(ConceptEvaluator__BehaviorDescriptor.java:86)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invokeSpecial(BaseBHDescriptor.java:357)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invokeVirtual(BaseBHDescriptor.java:316)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invoke(BaseBHDescriptor.java:240)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.SMethodImpl.invoke0(SMethodImpl.java:151)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.core.aspects.behaviour.SMethodImpl.invoke(SMethodImpl.java:126)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at com.mbeddr.mpsutil.interpreter.generator.template.main.QueriesGenerated.propertyMacro_GetValue_3_0(QueriesGenerated.java:86)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at com.mbeddr.mpsutil.interpreter.generator.template.main.QueriesGenerated$PVQ.evaluate(QueriesGenerated.java:729)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.template.DefaultQueryExecutionContext.evaluate(DefaultQueryExecutionContext.java:122)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateNode$PropertyMacro.expand(TemplateNode.java:333)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateNode.apply(TemplateNode.java:107)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.applyTemplate(TemplateProcessor.java:131)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.applyTemplate(TemplateProcessor.java:141)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.applyTemplate(TemplateProcessor.java:141)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.applyTemplate(TemplateProcessor.java:141)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.applyTemplate(TemplateProcessor.java:141)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.applyTemplate(TemplateProcessor.java:141)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.applyTemplate(TemplateProcessor.java:141)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.applyTemplate(TemplateProcessor.java:141)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.applyTemplate(TemplateProcessor.java:141)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.applyTemplate(TemplateProcessor.java:141)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.applyTemplate(TemplateProcessor.java:141)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.applyTemplate(TemplateProcessor.java:141)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateProcessor.apply(TemplateProcessor.java:106)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateContainer.processRuleConsequence(TemplateContainer.java:81)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.interpreted.TemplateReductionRuleInterpreted.apply(TemplateReductionRuleInterpreted.java:74)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.template.DefaultQueryExecutionContext.applyRule(DefaultQueryExecutionContext.java:219)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.TemplateExecutionEnvironmentImpl.tryToReduce(TemplateExecutionEnvironmentImpl.java:497)
[... cut ...]
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at jetbrains.mps.generator.impl.GenerationTaskPool$ModelReadAdapter.run(GenerationTaskPool.java:156)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
[07:08:14]W:		 [:code:buildLanguages]  [generate] 	at java.base/java.lang.Thread.run(Thread.java:834)
```

This has been puzzling me for a whilw and I am not sure how to investigate this further

sergej-koscejev, 2019-11-25:
1) https://youtrack.jetbrains.com/issue/MPS-30286 is a bug I reported about NPE being thrown from BasConcept.getPresentation() if the concept isn't loaded.
2) I think I fixed this part of the code in mbeddr.core in commit f12ef509ade78c50ab2fdf483e676dcc571fc83a in both master and 2019.1 version.

ftomassetti, 2019-11-25:
Unfortunately I am using the 2019.2 version. I have updated the dependencies this morning and I am still experiencing the bug

ftomassetti, 2019-11-25:
Thank you for the pointer!
For now I have created my own sub-concept of TypeChildConstraint where I override getPresentation. In my case it should work as a workaround

closed 2023-02-02 by alexanderpann