mbeddr/mbeddr.core #2109: Attribute open source packages used

eugenschindler, 2019-12-10:
Some of the open source packages that are used in mbeddr.platform (e.g. CSV reading library, XML library, etc.) have to be mentioned/attributed due to their license. It seems that mbeddr.platform doesn't do this at all or only partly. How do we go about adding proper attributions to all the software used? I think one file with a listing of all the open source packages used could already be enough.

slisson, 2019-12-11:
MPS seems to just have a folder with all the licenses (https://github.com/JetBrains/MPS/tree/master/license), that is also part of the binary distribution.
We could place the license file next to the jar and also package it with the plugin.

slisson, 2019-12-11:
There is also an about.txt (https://github.com/JetBrains/MPS/tree/master/about.txt) in the root folder that lists all the authors of the libraries. I guess we need such a file for each packaged plugin. Or is the description text of the plugin the correct place?

dbinkele, 2022-02-23:
@slisson @eugenschindler Can someone update the status of this ticket?!

arimer, 2024-03-07:
@slisson do we already place the missing licence files in the mentioned folder or/and with the plugin already?

eugenschindler, 2024-03-07:
I think this is something that @alexanderpann can answer?

alexanderpann, 2024-03-07:
Seems doable. There was recently a full scan for all open-source libraries in this repository done by another customer including the used licenses. Maybe we could get the list from them, I could ask.