mbeddr/mbeddr.core #2175: mbeddr.doc: DocumentationDescriptor cannot be resolved

arimer, 2021-05-17:
Adding a documenation aspect to your language and an empty `Document` http://127.0.0.1:63320/node?ref=r%3A5f7188a9-e7b4-4a2e-bef9-38d2cf379fdc%28com.mbeddr.doc.structure%29%2F3350625596579911728 leads to a compilation error. 

It looks like the Document needs to at least reference one `Concept` of the structure model of the corresponding language to be able to compile successfully.

eugenschindler, 2021-06-22:
To be a bit more precise: one needs to add a DocumentedConceptAnnotation to a section in order to make it have no compiler error. This can be done as follows:
* Create a section
* Stand on the section and invoke the /Add reference to documented concept/ intention
* Select a concept in the placeholder in the annotation above the intention that says /Documents .../

I tested and it also works with the /Add reference to documented property/ intention.

Thanks to @kbirken for the hints.
Maybe it's good to have a look at this, since more people are getting into this "undocumented feature" of the doc language?

If it is not possible to solve the issue timely, maybe it would be possible to make a mitigation? A very simple improvement could be to simply put a document model into the documentation aspect upon its creation, with a section and in it the hint text "In order to prevent compiler errors, execute "Add reference to documented concept/property" on this section and point to a concept of the language to which this documentation aspect belongs".

eugenschindler, 2021-06-24:
For making this piece of information better accessible to search engines: the error that you get in this specific situation is something along the lines of:
[jetbrains.mps.make.CompilationErrorsHandler] MbeddrDocLangTest\Language.java : The import MbeddrDocLangTest.documentation.DocumentationDescriptor cannot be resolved<br>[jetbrains.mps.make.CompilationErrorsHandler] MbeddrDocLangTest\Language.java : DocumentationDescriptor cannot be resolved to a type

alexanderpann, 2025-03-25:
I am going for the solution with a quick fix. Another workaround is to add the `com.mbeddr.doc.aspect` as an language engaged on generation in the documentation model.

closed 2025-04-07 by alexanderpann