mbeddr/mbeddr.core #2196: Exceptions using documentation

dbinkele, 2021-09-22:
A Stackoverflow happens in case there is documentation for the concept and the node.

java.lang.StackOverflowError
	at jetbrains.mps.lang.smodel.generator.smodelAdapter.MutableChildrenList.children(MutableChildrenList.java:44)
	at jetbrains.mps.lang.smodel.generator.smodelAdapter.AbstractMutableChildrenList.childrenIterator(AbstractMutableChildrenList.java:233)
	at jetbrains.mps.lang.smodel.generator.smodelAdapter.AbstractMutableChildrenList$1.<init>(AbstractMutableChildrenList.java:36)
	at jetbrains.mps.lang.smodel.generator.smodelAdapter.AbstractMutableChildrenList.iterator(AbstractMutableChildrenList.java:34)
	at jetbrains.mps.internal.collections.runtime.AbstractCollectionSequence.iterator(AbstractCollectionSequence.java:133)
	at jetbrains.mps.internal.collections.runtime.impl.FilteringSequence$FilteringIterator.init(FilteringSequence.java:56)
	at jetbrains.mps.internal.collections.runtime.impl.FilteringSequence$FilteringIterator.hasNext(FilteringSequence.java:31)
	at jetbrains.mps.internal.collections.runtime.IterableUtils.isEmpty(IterableUtils.java:52)
	at jetbrains.mps.internal.collections.runtime.AbstractSequence.isEmpty(AbstractSequence.java:179)
	at jetbrains.mps.lang.smodel.generator.smodelAdapter.IAttributeDescriptor$AttributeDescriptor.get(IAttributeDescriptor.java:165)
	at com.mbeddr.core.base.behavior.IDocumentable__BehaviorDescriptor.getDocNodeOrNull_id11K_5nNeRli(IDocumentable__BehaviorDescriptor.java:50)
	at com.mbeddr.core.base.behavior.IDocumentable__BehaviorDescriptor.invokeSpecial0(IDocumentable__BehaviorDescriptor.java:77)
	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invokeSpecial(BaseBHDescriptor.java:358)
	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invokeVirtual(BaseBHDescriptor.java:317)
	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invoke(BaseBHDescriptor.java:241)
	at jetbrains.mps.core.aspects.behaviour.SMethodImpl.invoke0(SMethodImpl.java:151)
	at jetbrains.mps.core.aspects.behaviour.SMethodImpl.invoke(SMethodImpl.java:126)
	at com.mbeddr.doc.aspect.ui.plugin.TabbedDocViewer_Tool.showDocumentation(TabbedDocViewer_Tool.java:216)
	at com.mbeddr.doc.aspect.ui.plugin.TabbedDocViewer_Tool.showDocumentation(TabbedDocViewer_Tool.java:204)
	at com.mbeddr.doc.aspect.ui.plugin.DocumentationHandler$1.run(DocumentationHandler.java:65)
	at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:81)
	at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:56)
	at com.mbeddr.doc.aspect.ui.plugin.DocumentationHandler.showDoc(DocumentationHandler.java:38)
	at com.mbeddr.doc.aspect.ui.plugin.TabOnShowListenerAdapter.ancestorAdded(TabOnShowListenerAdapter.java:30)
	at java.desktop/javax.swing.AncestorNotifier.fireAncestorAdded(AncestorNotifier.java:86)
	at java.desktop/javax.swing.AncestorNotifier.propertyChange(AncestorNotifier.java:223)
	at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:341)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:333)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:266)
	at java.desktop/java.awt.Component.firePropertyChange(Component.java:8754)
	at java.desktop/javax.swing.JComponent.addNotify(JComponent.java:4792)
	at java.desktop/java.awt.Container.addImpl(Container.java:1146)
	at java.desktop/java.awt.Container.add(Container.java:997)
	at com.intellij.openapi.wm.impl.content.ToolWindowContentUi.ensureSelectedContentVisible(ToolWindowContentUi.java:226)
	at com.intellij.openapi.wm.impl.content.ToolWindowContentUi$1.selectionChanged(ToolWindowContentUi.java:124)
	at jdk.internal.reflect.GeneratedMethodAccessor67.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at com.intellij.util.EventDispatcher.dispatchVoidMethod(EventDispatcher.java:123)
	at com.intellij.util.EventDispatcher.lambda$createMulticaster$1(EventDispatcher.java:86)
	at com.sun.proxy.$Proxy77.selectionChanged(Unknown Source)
	at com.intellij.ui.content.impl.ContentManagerImpl.fireSelectionChanged(ContentManagerImpl.java:564)
	at com.intellij.ui.content.impl.ContentManagerImpl.addSelectedContent(ContentManagerImpl.java:389)
	at com.intellij.ui.content.impl.ContentManagerImpl$1.run(ContentManagerImpl.java:473)
	at com.intellij.ui.content.impl.ContentManagerImpl.setSelectedContent(ContentManagerImpl.java:489)
	at com.intellij.ui.content.impl.ContentManagerImpl.setSelectedContentCB(ContentManagerImpl.java:442)
	at com.intellij.ui.content.impl.ContentManagerImpl.setSelectedContentCB(ContentManagerImpl.java:432)
	at com.intellij.ui.content.impl.ContentManagerImpl.setSelectedContentCB(ContentManagerImpl.java:506)
	at com.intellij.ui.content.impl.ContentManagerImpl.setSelectedContent(ContentManagerImpl.java:511)
	at com.mbeddr.doc.aspect.ui.plugin.TabbedDocViewer_Tool.showDocumentation(TabbedDocViewer_Tool.java:242)
	at com.mbeddr.doc.aspect.ui.plugin.TabbedDocViewer_Tool.showDocumentation(TabbedDocViewer_Tool.java:204)
	at com.mbeddr.doc.aspect.ui.plugin.DocumentationHandler$1.run(DocumentationHandler.java:65)
	at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:81)
	at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:56)
	at com.mbeddr.doc.aspect.ui.plugin.DocumentationHandler.showDoc(DocumentationHandler.java:38)
	at com.mbeddr.doc.aspect.ui.plugin.TabOnShowListenerAdapter.ancestorAdded(TabOnShowListenerAdapter.java:30)
	at java.desktop/javax.swing.AncestorNotifier.fireAncestorAdded(AncestorNotifier.java:86)
	at java.desktop/javax.swing.AncestorNotifier.propertyChange(AncestorNotifier.java:223)
	at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:341)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:333)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:266)
	at java.desktop/java.awt.Component.firePropertyChange(Component.java:8754)
	at java.desktop/javax.swing.JComponent.addNotify(JComponent.java:4792)
	at java.desktop/java.awt.Container.addImpl(Container.java:1146)
	at java.desktop/java.awt.Container.add(Container.java:997)
	at com.intellij.openapi.wm.impl.content.ToolWindowContentUi.ensureSelectedContentVisible(ToolWindowContentUi.java:226)
	at com.intellij.openapi.wm.impl.content.ToolWindowContentUi$1.selectionChanged(ToolWindowContentUi.java:124)
[... 932 lines cut ...]
	at com.intellij.util.EventDispatcher.lambda$createMulticaster$1(EventDispatcher.java:86)
	at com.sun.proxy.$Proxy77.selectionChanged(Unknown Source)
	at com.intellij.ui.content.impl.ContentManagerImpl.fireSelectionChanged(ContentManagerImpl.java:564)
	at com.intellij.ui.content.impl.ContentManagerImpl.addSelectedContent(ContentManagerImpl.java:389)
	at com.intellij.ui.content.impl.ContentManagerImpl$1.run(ContentManagerImpl.java:473)
	at com.intellij.ui.content.impl.ContentManagerImpl.setSelectedContent(ContentManagerImpl.java:489)
	at com.intellij.ui.content.impl.ContentManagerImpl.setSelectedContentCB(ContentManagerImpl.java:442)
	at com.intellij.ui.content.impl.ContentManagerImpl.setSelectedContentCB(ContentManagerImpl.java:432)
	at com.intellij.ui.content.impl.ContentManagerImpl.setSelectedContentCB(ContentManagerImpl.java:506)
	at com.intellij.ui.content.impl.ContentManagerImpl.setSelectedContent(ContentManagerImpl.java:511)
	at com.mbeddr.doc.aspect.ui.plugin.TabbedDocViewer_Tool.showDocumentation(TabbedDocViewer_Tool.java:242)
	at com.mbeddr.doc.aspect.ui.plugin.TabbedDocViewer_Tool.showDocumentation(TabbedDocViewer_Tool.java:204)
	at com.mbeddr.doc.aspect.ui.plugin.DocumentationHandler$1.run(DocumentationHandler.java:65)
	at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:81)
	at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:56)
	at com.mbeddr.doc.aspect.ui.plugin.DocumentationHandler.showDoc(DocumentationHandler.java:38)
	at com.mbeddr.doc.aspect.ui.plugin.TabOnShowListenerAdapter.ancestorAdded(TabOnShowListenerAdapter.java:30)
	at java.desktop/javax.swing.AncestorNotifier.fireAncestorAdded(AncestorNotifier.java:86)
	at java.desktop/javax.swing.AncestorNotifier.propertyChange(AncestorNotifier.java:223)
	at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:341)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:333)
	at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:266)
	at java.desktop/java.awt.Component.firePropertyChange(Component.java:8754)
	at java.desktop/javax.swing.JComponent.addNotify(JComponent.java:4792)
	at java.desktop/java.awt.Container.addImpl(Container.java:1146)

dbinkele, 2021-09-22:
The follwing situation causes the problem :

[image]

alexanderpann, 2022-04-27:
Closing because it was fixed in PR #2197. Please reopen if problem still exists.

closed 2022-04-27 by alexanderpann