mbeddr/mbeddr.core #2219: Layout problems with SqrtExpression

josefmueller, 2022-05-30:
There is a layout problem for the combination of SqrtExpression, PowerExpression and LocalVarRef. Using ParensExpression does not fix it.

Example:
[image]

eugenschindler, 2022-07-11:
I can confirm this also on MPS 2021.1 with mbeddr tutorial HelloMath module:
project: tutorial (/home/eugen/repo/external/mbeddr.core/code/applications/tutorial)
module:  mbeddr.tutorial.main
model:   mbeddr.tutorial.main.math
root:    HelloMath [ImplementationModule]
node:    midnight1 [Function]
url:     http://127.0.0.1:63320/node?ref=r%3A6a63d759-341b-4a55-b0b8-859e3707a9ff%28mbeddr.tutorial.main.math%29%2F7040583391080129354

alexanderpann, 2022-07-20:
I think the bug is in the celllayout language in mps-extensions. When I deactivate the layout interceptor, the rendering works. When I surround the `CellModel_RefCell` in the editor of ArgumentRef/LocalVarRef with a collection, everything also works, but that's not a proper fix. There must be a reason, why the root cell of the editor has to be a collection in the editor. Maybe @slisson has a clue why this is happening.

alexanderpann, 2024-07-14:
This issue looks fixed in the maintenance/mps2022 branch through one of the latest fixes in the celllayout language in MPS-Extensions. Please reopen when you find an edge case where it doesn't work.

closed 2024-07-14 by alexanderpann